FreeCAD/FreeCAD #14465: [Core][Part Design][Sketch] Making a sketch on another PartFace has issues.

luisDias89, 2024-06-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Both the plane on which the sketch was created did not coincide with the face of the other object, and the copy does not match the face of the object either.

Video with problem:
https://github.com/FreeCAD/FreeCAD/assets/48370099/b2dbb79d-683d-49b2-b94a-ad3a3c2a5312

Exemple of good Work:
https://github.com/FreeCAD/FreeCAD/assets/48370099/11decb3f-244c-4793-b221-c1828087318c

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Portugal (pt_PT)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-03:
Does this work in the stable 0.21.2 version?

FlachyJoe, 2024-06-03:
@Syres916 not ok, the Part placement is not taken into account even in
```
OS: Debian GNU/Linux trixie/sid (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.2.
Build type: Release
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
```
[image]

Another bug is the shape copy. 
Why this cube is still visible when Part is hidden? Because it was taken as the Base for the Pad and fused with it. Not good.
[image]

Syres916, 2024-06-03:
Apologies I'd not noticed the Part container as I never use them my test file doesn't have one.

luisDias89, 2024-06-03:
@maxwxyz Test with stabel version DONE

Video with test: 

https://github.com/FreeCAD/FreeCAD/assets/48370099/551a650c-a2aa-4793-bafa-fa0b55e8dd70

Tests completed, I found some bugs and differences.
1º - The plan allocation in space is okay.
2º It never brings the geometric references as it does in the new version.
3º The independent copy does not work; it is bugged.

FEA-eng, 2025-07-17:
@maxwxyz This is still relevant. Can you reopen the issue?

Steps to reproduce:
1. Create a Part and Body inside it. Add a Box primitive to it.
2. Edit the Part container's placement to move it a bit.
3. Deactivate the Part and create a new Body.
4. With the new Body active, select a face of the Box and create a new sketch.
5. In the pop-up window choose the second or third option. The former will create a shapebinder that ignores the placement of the Box while the latter will locate the sketch plane incorrectly.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42564 (Git)
Build date: 2025/07/14 18:51:42
Build type: Release
Branch: main
Hash: dc2f2374b6ab3e3381653fad125b1b619bb49f7c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```