FreeCAD/FreeCAD #14466: [Assembly] : Joints between a vertex/edge and a line or path of a sketch.

luisDias89, 2024-06-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've been analyzing, and I still can't figure out how to create a virtual path for a mechanical system in the new assembly. For instance, it involves adding a sketch that outlines the path it should follow. At the end, we can hide the sketch, and a given object only follows that path. I'll provide a well-elaborated example to demonstrate

Exemple:

https://github.com/FreeCAD/FreeCAD/assets/48370099/51f76797-844c-4c9a-9056-d595a03cd234

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Portugal (pt_PT)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-03:
@PaddleStroke FYI

PaddleStroke, 2024-06-11:
This feature is not implemented yet. (I'm not sure but there might already be an opened issue for this FR.)

PaddleStroke, 2024-06-11:
Ah yes it's duplicate of https://github.com/FreeCAD/FreeCAD/issues/13388

maxwxyz, 2024-06-11:
Duplicate of #13388

closed 2024-06-11 by maxwxyz