FreeCAD/FreeCAD #14471: QuantitySpinBox loses value - (reverting back to previous set rawValue)

akapelrud, 2024-06-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Example of behavior:
```Python
>>> qi = FreeCADGui.UiLoader().createWidget('Gui::QuantitySpinBox')
>>> qi.setProperty('rawValue', 3.5)
True
>>> qi.show()
>>> qi.setProperty('value', FreeCAD.Units.Quantity('2 V'))
```
The value shown in the spinbox is updated to the correctly scaled units, i.e. "2000000,00 mm^2*kg/(s^3*A)", after the 4th command, but if I select the spinbox window and go back to the main freecad window the spinbox reverts back to displaying '3,5', i.e. the old rawValue that was set first.

This bug is also apparent in FEM electrostatic potential task panel, `task_constraint_electrostaticpotential.py`, where the potential always displays "0,0" on my system, even though the `_initParamWidget(self)` member of that task panel sets the potential value from the saved Quantity object.

It could be that this is a locale-dependent and/or platform-dependent bug where some function for the quantity validation in the spinbox fails silently.
Note. I do not see this bug when running the 0.20 version that is available for Debian through WSL2 on Win10_64.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37556 (Git)
Build type: Release
Branch: main
Hash: df68858765a08ed5a058c743e137bb38f57eec34
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Norwegian Bokmal/Norway (nb_NO)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

akapelrud, 2024-06-03:
In `QuantitySpinBox.cpp`, `void setValue(const Base::Quantity& value)` calls `updateText(value)` which again delegates to `updateEdit()`. This function only updates the LineEdit.

Verification of the lineedit is apparently enforced in `focusOutEvent` which calls `validateInput()` which leads to a call to the regex-heavy function `validateAndInterpret()`. I'm guessing that function does not return a `QValidator::Acceptable` as it should, resulting in a reset of the text.

akapelrud, 2024-06-03:
Digging deeper, `validateAndInterpret()` calls `parseString()` which catches a `Base::Exception&` thrown by `ExpressionParser::parse()`. The error stems from the parser giving up in the line:

`int result = ExpressionParser::ExpressionParser_yyparse ();`

which leads to one of the inner circles of Bison hell. So the question is, is the expression parser locale aware? Will it allow numbers where (comma) ',' is the decimal point?

xtemp09, 2024-06-03:
Something tricky with this. It works as expected on _Windows_; the fourth command does not work on _Linux_.

I have set '.' as a decimal separator on both _Windows_ and _Linux_.

<details><summary>Windows 10 build 19044</summary>

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git)
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
```

</details>

<details><summary>Ubuntu 22.04.4 LTS (KDE/plasma)</summary>

```ruby
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Debug
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.10.12, Qt 6.7.1, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

```ruby
LANG=en_US.UTF-8
LANGUAGE=en_US:
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=ru_RU.UTF-8
LC_TIME=ru_RU.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=ru_RU.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=ru_RU.UTF-8
LC_NAME=ru_RU.UTF-8
LC_ADDRESS=ru_RU.UTF-8
LC_TELEPHONE=ru_RU.UTF-8
LC_MEASUREMENT=ru_RU.UTF-8
LC_IDENTIFICATION=ru_RU.UTF-8
LC_ALL=
```

</details>

akapelrud, 2024-06-03:
I enforced a substitute of ',' (comma) with '.' before the validation for a quick test, and that didn't change anything - the ExpressionParser still gave up.

Next, playing with the expression editor directly in the gui for the ConstraintElectrostaticPotential case, I see that the compound unit `mm^2*kg/(s^3*A)`, which is equal to 1 uV is not accepted by the parser:
[image]

On the other hand, if I enclose the numerator part of the unit in parenthesis, it gets parsed correctly:
[image]

This seems to stem from a bug with the unit parsing of the ExpressionParser, not the value parsing as I first thought.

marioalexis84, 2024-06-03:
@maxwxyz  Could you remove the FEM label?
This is a general problem with expressions in widgets when trying to use the `*` character.

akapelrud, 2024-06-26:
This bug can be demonstrated by adding a test to 
```C++
EXPECT_NO_THROW(App::ExpressionParser::parse(this_obj(), "1 mm*kg")) << "should parse a simple compound unit";
```
to the `TEST_F`block in the gtest file `tests/src/App/ExpressionParser.cpp`.

For some reason, I am not able to set `App::ExpressionParser::yydebug=1` (bison debug variable) from within this gtest block. Does anyone have any tips on debugging the bison expressionparser in FreeCAD?

akapelrud, 2024-06-27:
Nevermind, I needed to manually add an `AppExport` (i.e. dllexport/dllimport) to the yydebug variable in `src/App/ExpressionParser.tab.{c,h}`

Activating bison tracing ([bison manual#tracing](https://www.gnu.org/software/bison/manual/bison.html#Tracing)) through
```c++
App::ExpressionParser::yydebug=1;
EXPECT_NO_THROW(App::ExpressionParser::parse(this_obj(), "1 mm*kg")) << "should parse a simple compound unit";
App::ExpressionParser::yydebug=0;
```
and redirecting `stderr` output (debug prompt in MSVC)
```
[...]\FreeCAD\build\bin>Tests_run.exe --gtest_filter=ExpressionParserTest* 2> bison_debug_stderr.txt
```
I get the trace:
```
[bison_debug_stderr.txt](https://github.com/user-attachments/files/16012349/bison_debug_stderr.txt)
```

akapelrud, 2024-06-27:
The handling of the expression "1 mm*kg" is starting at line 293 `Starting parse` in the previously linked trace file.

Using the nomenclature from [ExpressionParser.y](https://github.com/FreeCAD/FreeCAD/blob/main/src/App/ExpressionParser.y), 
"1 mm" is parsed to the stack `num unit_exp` and prematurely reduces this further to `unit_num` and subsequently to `exp`.

At this point we see that the reduction of "num unit_exp" to `unit_num` was an error because the next token is `*` followed by `UNIT`. Further lookahead and a shift here should have resulted in the stack `num unit_exp '*' unit_exp` being reduced to `num unit_exp`, which in this case would have completed the parse.

There is no rule for parsing the stack `exp '*' unit_exp`, so the parser throws an error.

As such, there is a precedence error and/or rule order error in `ExpressionParser.y`.

akapelrud, 2024-06-27:
@maxwxyz and @marioalexis84; changing the operator precedence fixes this bug:
```
diff --git a/src/App/ExpressionParser.y b/src/App/ExpressionParser.y
index 976d3eaa20..6cc9007d89 100644
--- a/src/App/ExpressionParser.y
+++ b/src/App/ExpressionParser.y
@@ -68,8 +68,8 @@ std::stack<FunctionExpression::Function> functions;                /**< Function
      %precedence EQ NEQ LT GT GTE LTE
      %precedence ':'
      %left MINUSSIGN '+'
-     %left '*' '/' '%'
      %precedence NUM_AND_UNIT
+     %left '*' '/' '%'
      %left '^'
      %precedence NEG
      %precedence POS
```

All gtests still pass and my the test I used for debugging above parses correctly. This also fixes the part of the bug relating to the FEM electrostatic potential task panel I initially reported on.

FEA-eng, 2025-10-18:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/14 15:14:51
Build type: Release
Branch: main
Hash: 41ce37233d64c81c2f105de085635d4dcd669943
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```