FreeCAD/FreeCAD #14473: FATAL:zygote_host_impl_linux.cc(208)] Check failed

Delan168, 2024-06-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD doesn't open: splashscreen briefly appears, then nothing. When launched from the terminal (by just typing `freecad`) following error is raised
`FATAL:zygote_host_impl_linux.cc(208)] Check failed: . : Aucun fichier ou dossier de ce nom (2)`
(roughly translated by "No file or folder with that name")

I use Manjaro, fully updated. This problem doesn't appear to be specific to FreeCAD since I saw the same kind of error online for other software, yet it only appears on my side with FreeCAD.
Full terminal output below:
```
user@host ~> freecad 
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

[4069:4069:0603/145738.400958:FATAL:zygote_host_impl_linux.cc(208)] Check failed: . : Aucun fichier ou dossier de ce nom (2)
fish: Job 1, 'freecad' terminated by signal SIGTRAP (Déroutement de suivi/point d'arrêt)
```

### Full version info

```shell
Not possible, no GUI launches...
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Delan168, 2024-06-06:
It works fine! Thank you for the hotfix, I'll check if it's stable enough for my occasional use :)
Any idea of what causes the issue in the v0.21?

`OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git) AppImage
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2`

Delan168, 2024-06-09:
Hello @maxwxyz , the dev version is way too unstable to be usable :/ (for example, the pop-up to save a file crashes systematically - what is the best place to report this bug?).
Do you have an idea on how to solve the issue for the stable release?

maxwxyz, 2024-06-09:
create a new issue for all bugs in dev.
The stable version will not get any bugfixes, as we are currently in feature freeze for the next stable release.

closed 2024-06-09 by maxwxyz

Delan168, 2024-06-09:
Issue solved with v21.2: Looks like the AppImage is somehow flawed for some OS, installing from flatpak did the job.