FreeCAD/FreeCAD #14479: Sketcher: There is no way to cancel modificatinos

pierreporte, 2024-06-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The sketcher workbench lacks a way to cancel what has been done in the opened sketch:

When making modifications to a sketch that are unwanted in the actual model (to make tests or things like that), there is no way to discard them but deleting them by hand. Even if it’s just moving dimensions. A recompute is then triggered.

When a sketch is opened and nothing is done before exiting, the model is recomputed.

The sketcher workbench lacks a way to cancel what has been done in the opened sketch.

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git) AppImage
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * BillOfMaterials-WB 0.0.7
  * CosmeticThread3D
  * OpenTheme 2024.4.20
  * Curves 0.6.35
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.13
```

### Subproject(s) affected?

Sketcher

### Anything else?

https://forum.freecad.org/viewtopic.php?t=88022

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-03:
Automatic recompute can be disabled. Does Undo not work as workaround?

pierreporte, 2024-06-03:
Undoing doesn’t work using the following procedure:

1. Create a sketch with a circle which center is constrained with the origin.
2. Exit the sketch and enter it again.
3. Add something to the sketch.
4. Click undo: the added object is removed.
5. Undo again: the circle center is no longer coincident with the origin. The sketch is still open.
6. 3rd undo: the circle is removed and the sketch is still open.
7. 4th undo: the sketch is removed.

PaddleStroke, 2024-06-13:
There is another issue about this recently opened that asked for a 'Cancel' button next to the Close button.

maxwxyz, 2024-06-13:
https://github.com/FreeCAD/FreeCAD/issues/14634 this one is related but just about exiting with Esc.

samohd, 2024-08-30:
The irritating result of this is this perspective...Open a document. In Part Design, double-click a sketch to view it. The document is now flagged as modified and will prompt to save when you close it. This is silly. It defeats the whole point of backup files... Since just looking at a document will potentially result in the document files and backups being replaced.

I've even seen document A flagged as changed, when I was working in document B in another tab.

No idea why a recompute would be considered a change. Regardless, the default state should not be to make backups useless. :-)

pierreporte, 2024-08-30:
Related: #6290

pgilfernandez, 2024-09-21:
I do agree a "cancel" button is needed and somehow it will add some equilibrium to the whole app as other Tasks already have the cancel button (Transform for instance...)

lmutt, 2024-10-09:
I totally agree with this issue. A cancel button would be useful. As said before, just opening a sketch to inspect it and closing it causes the whole project to be recalculated, and the recalculation is often time-consuming. I think that manually disabling the automatic recalculation every time you need to open a sketch just to inspect it is not the best option. I hope this feature will be implemented someday :)