FreeCAD/FreeCAD #14482: [UI @ NEW INTELIGENCE FEATURE] : menu inteligente que deteta quais as principais operações que podem ser realizadas

luisDias89, 2024-06-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Add an intelligent menu that, when the user clicks on a vertex, plane, edge, etc., displays the main operations available for the selected item. In other words, the menu is interactive and always depends on what is selected.
The menu appears shaded, and only becomes active if the user moves the mouse over it. If the mouse moves in the opposite direction, it disappears.
This menu ensures that 90% of the frequently used features appear directly in it, increasing productivity. The interaction with the user is more fluid because it provides an overview of what can be used for the selected item.

It doesn't have to be identical, but an approximation of the idea.

I made a video with the exemple:

https://github.com/FreeCAD/FreeCAD/assets/48370099/9ddc8d53-62fd-4d03-b2f8-c17cba972ed2

Images exemples:
[image]

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Portugal (pt_PT)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Reqrefusion, 2024-06-03:
I love this feature. Most of the time it does not work properly even in other programs.

maxwxyz, 2024-06-03:
This is a very broad request not describing a concise problem.
It's a duplicate of https://github.com/FreeCAD/FreeCAD/issues/11413 for the Sketcher WB.