FreeCAD/FreeCAD #14486: TechDraw: Pan resets dimension position

maxwxyz, 2024-06-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When rearranging dimensions in TechDraw moving on the page (pan) resets the dimension position. I'm using CAD navigation mode.
This also happens with old FCStd files created with 0.21.2
Could this be introduced by https://github.com/FreeCAD/FreeCAD/pull/14359

https://github.com/FreeCAD/FreeCAD/assets/6246609/aaf117c8-0690-4c62-a31e-bed2ba38bff5

@PaddleStroke FYI

[test.zip](https://github.com/user-attachments/files/15538573/test.zip)

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36729 (Git)
Build type: Release
Branch: main
Hash: 6ca35709ddc7f95fb84eef4c24973dc489e5acde
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * CfdOF 1.25.17
  * Curves 0.6.37
  * dodo 1.0.1
  * fasteners 0.5.21
  * freecad.gears 1.2.0
  * OpenTheme 2024.5.16
  * OpticsWorkbench 1.0.17
  * Rocket 3.3.0
  * sheetmetal 0.4.16
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-06-04:
@maxwxyz can you confirm that this did no happen with 0.21?

maxwxyz, 2024-06-04:
@PaddleStroke the file does not display correctly with the 0.21.2 stable but I can recreate the view and dimension and I cannot reproduce the dimension reset while moving around the page (pan).

PaddleStroke, 2024-06-13:
I cannot reproduce this problem anymore. Can you please rebase and check if it is gone on your side too?

maxwxyz, 2024-06-13:
I cannot reproduce either!

closed 2024-06-13 by maxwxyz