FreeCAD/FreeCAD #14488: [ASSEMBLY] : LimitDistance is broken, with no sense.

luisDias89, 2024-06-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The limit tool is overlapping with the other "Joins" and causing strange behavior. Here is a video demonstrating the issue:
https://github.com/FreeCAD/FreeCAD/assets/48370099/e2901bfa-4268-4bfd-bef9-6c9bb496c81e

Something simple is being made complicated. The limits should be in relation to two planes and a point and should be part of the distance join.

The limits can be set between:

Two vertices
A vertex and a point
Two circles
A circle and a point
And so on...

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Portugal (pt_PT)
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

howie-j, 2024-06-03:
Why are you using two distance constraints in the first place? You are overconstraining the assembly. Use the slider joint only, and use the limits in that joint.

luisDias89, 2024-06-03:
> Why are you using two distance constraints in the first place? You are overconstraining the assembly. Use the slider joint only, and use the limits in that joint.

In response to your question, here's a video showing that everything is still wrong...

https://github.com/FreeCAD/FreeCAD/assets/48370099/63fb0f37-8021-4209-ace0-406be3e81e4d

luisDias89, 2024-06-03:
I'm trying to port a CNC project from one software to FreeCAD, and I haven't stopped reporting bugs on GIT. It's impossible to work with this software as it is.
PS: But I am persistent.

howie-j, 2024-06-03:
Remember you are using freecad, not solidworks. The assembly workbench is inspired more by onshape than solidworks. If you are struggling to understand the assembly workbench, try watching an onshape assembly tutorial.

Here is your assemly file fixed:
[WorkSpaceTestFreecad_Fixed.zip](https://github.com/user-attachments/files/15540793/WorkSpaceTestFreecad_Fixed.zip)

[Screencast from 2024-06-04 01-40-14.webm](https://github.com/FreeCAD/FreeCAD/assets/82042580/a45dee39-dd88-4c51-bdd3-55a294cc77dd)

luisDias89, 2024-06-03:
@howie-j Regarding "overconstraining," the technique I used is not overconstraining. It's the technique used in top-tier software like SOLIDWORKS. There are no slicers, only primitives. In fact, I would delete that slicer if it were up to me.

I have assemblies with more than 1000 parts working, and I only use primitives.
[image]

luisDias89, 2024-06-04:
@howie-j I managed to do it as well, but it's very cumbersome and complicated because, after that, I still have to rotate the object 180 degrees to get it straight.

So, it's necessary to have a vertex that coincides with the other. The question is, I have assemblies with a 2mm gap between the slicer and two coincident points. How is this going to work? Or, I even have parts where the slicing surface is rough.

For example, does this ensure the gap?
[image]

luisDias89, 2024-06-04:
@PaddleStroke, Could you please analyze and respond to this?

PaddleStroke, 2024-06-04:
In your first video, you make 2 distance joints + 1 slider joint, when you should need only 1 slider joint. So the assembly is overconstrained resulting in the bad behavior.

So this issue should be closed.

However you raise a valid point with your latest picture. I guess the correct solution would be to enable the user to set an offset on X/Y on joint creation. If you can create an issue about this it would be great.

luisDias89, 2024-06-06:
@PaddleStroke  In response to your question, yes, it could be 2 distances of 1 mm and then just one Distance feature.
In my opinion, that's what should exist, just one Distance feature, because the distance can be between points, between planes, between lines, between lines and points, etc.

Notes:
From this issue, I'll open two. One because of what happened in the first video. There should be a routine that checks if adding a new Joint was breaking the previous ones, and a window would appear with the question: Is there a collision between Constraints? Do you want to add this and Suppress the broken Constraints, or cancel the addition?

And right after, I checked that there are still more situations, beyond the GAP, that the slider tool can't resolve