FreeCAD/FreeCAD #14492: Sketcher: There should be a preference to have Polyline and Line grouped together

FEA-eng, 2024-06-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

#14299 reverted the controversial change in the Sketcher which grouped Polyline and Line tools with Polyline being on the top. However, some users would still like to keep this grouping so a preference for that could be introduced, similar to the settings for the new Dimension tool.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37645 (Git)
Build type: Release
Branch: main
Hash: fc714933a7e2423d5fa4db640af549814061c42e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-04:
Maybe it should just be instead a better polyline tool to get rid on a single line tool.

FEA-eng, 2024-06-04:
> Maybe it should just be instead a better polyline tool to get rid on a single line tool.

I'm afraid that even then people will want to keep the Line tool so this should be the best compromise for now and then.

kaiwas, 2025-03-13:
In short - all these groupings are implemented in a completely inconvenient way. You have to constantly aim at the small triangle for the drop-down list and then click.
Maybe if the list dropped down when you hover the mouse and hold it over the icon for more than 1 second, it would be much better.

Fortunately, it is possible to make your own panel with the necessary buttons without any grouping.

closed 2025-03-17 by chennes