FreeCAD/FreeCAD #14498: BIM: IfcOpenShell Update command fails if IfcOpenShell already installed

furgo16, 2024-06-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. IfcOpenShell is already installed on the system via pip
1. Start FreeCAD
1. Activate the BIM Workbench
1. Go to the <kbd>Utils</kbd> > <kbd>IfcOpenShell update</kbd> menu entry
1. The command fails with the following traceback:

```python-traceback
14:56:09  Running the Python command 'IFC_UpdateIOS' failed:
Traceback (most recent call last):
  File "$HOME/dev/FreeCAD/build/Mod/BIM/nativeifc/ifc_openshell.py", line 53, in Activated
    comp = self.compare_versions(avail, version)
  File "$HOME/dev/FreeCAD/build/Mod/BIM/nativeifc/ifc_openshell.py", line 157, in compare_versions
    arr2 = [int(i) for i in arr2]
  File "$HOME/dev/FreeCAD/build/Mod/BIM/nativeifc/ifc_openshell.py", line 157, in <listcomp>
    arr2 = [int(i) for i in arr2]

invalid literal for int() with base 10: 'v0'
```

Note that this was with a local build. After adding some print statements, I noticed that the `get_current_version()` function fails because `get_pip_target_directory()` returns a path that the subsequent `pip` command does not understand:

```
Current version: v0.7.0-f7c03db75 # As returned by `ifcopenshell.version`
Vendor path: $HOME/.local/share/FreeCAD/Mod/../AdditionalPythonPackages/py310 # As returned by addonmanager_utilities.get_pip_target_directory()
Result: [] # After running: result = self.run_pip(["list", "--path", vendor_path])
Available version: 0.7.0.240521 # After running: avail = self.get_avail_version()
```

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37578 +9 (Git)
Build type: Unknown
Branch: Draft_Grid_Observer_clean
Hash: 01b2b370e9b7538eb20ab07dd20ff036e36f51e5
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Installed mods: 
  * OpenTheme 2024.5.3
  * parts_library
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-04:
@yorikvanhavre FYI

yorikvanhavre, 2024-06-06:
See https://github.com/FreeCAD/FreeCAD/issues/14499#issuecomment-2152244758

maxwxyz, 2024-06-06:
I'll close this then if the plan is to get rid of this

closed 2024-06-06 by maxwxyz