FreeCAD/FreeCAD #14499: BIM: disable

furgo16, 2024-06-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On sandboxed systems (e.g. snap and flatpak packages in Linux), `ifcopenshell` cannot be updated in principle. At this time, the command fails with a traceback. It might be best to either disable it, or to present a friendlier dialog telling the user that it cannot be updated.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37579 (Git) Snap 1039
Build type: Release
Branch: main
Hash: 77fc1de6c334b5bacbeb43f466f1c55253c37a32
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.5.16
  * OpenTheme.backup1716114997.8759558 2024.5.16 (Disabled)
  * Curves 0.6.36
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.2
  * Ondsel-Lens 2024.5.6.01
  * Assembly4 0.50.12
```

### Subproject(s) affected?

BIM/Arch

### Anything else?

```python-traceback
15:33:52  Running the Python command 'IFC_UpdateIOS' failed:
Traceback (most recent call last):
  File "/snap/freecad/1039/usr/Mod/BIM/nativeifc/ifc_openshell.py", line 48, in Activated
    version = self.get_current_version()
  File "/snap/freecad/1039/usr/Mod/BIM/nativeifc/ifc_openshell.py", line 125, in get_current_version
    result = self.run_pip(["list", "--path", vendor_path])
  File "/snap/freecad/1039/usr/Mod/BIM/nativeifc/ifc_openshell.py", line 110, in run_pip
    return utils.run_interruptable_subprocess(cmd)
  File "/snap/freecad/1039/usr/Mod/AddonManager/addonmanager_utilities.py", line 429, in run_interruptable_subprocess
    raise subprocess.CalledProcessError(

Command '['/snap/freecad/1039/bin/python3', '-m', 'pip', 'list', '--path', '$HOME/snap/freecad/common/Mod/../AdditionalPythonPackages/py310']' returned non-zero exit status 1.
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-04:
@yorikvanhavre FYI

yorikvanhavre, 2024-06-06:
I wonder in any case if we need to keep this command... With IfcOpenShell bundled in almost all packages, it makes no more sense. What about removing the command and adding it as a macro to the macros repo? So people can still install and use it if needed

furgo16, 2024-06-06:
> What about removing the command and adding it as a macro to the macros repo? So people can still install and use it if needed

I was thinking the same as I was filing the issues. I feel that sounds sensible. FreeCAD is available in 3 platforms. Setting Windows and MacOS aside, just for Linux alone there are more than 5 package formats it can be installed from. On top of that, IfcOpenShell can either be bundled, installed from source, via pip, conda and probably some other method I'm missing:

1. FreeCAD as AppImage package: IfcOpenShell is bundled <= official installation method
1. FreeCAD as Snap package: IfcOpenShell is bundled
1. FreeCAD as Debian package: as IfcOpenShell is not packaged as .deb, it probably has to be installed manually via pip or other methods
1. FreeCAD as flatpak package: I don't know. IfcOpenShell probably can be bundled, but I'm not sure if it's been done
1. FreeCAD as rpm package: I don't know, probably similar issue as the Debian package
1. FreeCAD as ArchLinux package: it seems IfcOpenShell is not added as a dependency: [https://aur.archlinux.org/packages/free ... =1#pkgdeps](https://aur.archlinux.org/packages/freecad-git?all_deps=1#pkgdeps)

The non-exhaustive list above is just to illustrate that trying to support all installation combinations might be a substantial development and maintenance effort. Perhaps not for the complexity of it, but rather for the sheer number of methods.

I would argue that if at least the official installation method for each of the platforms has IfcOpenShell bundled, removing the command would allow redirecting the time investment to more pressing areas of FreeCAD/BIM WB.

yorikvanhavre, 2024-06-25:
So basically our problem is mostly linux distributions (we can assume the flatpack should get fixed). For those indeed we should provide a way to install IfcOpenShell via pip.

I guess then what we need here is to ensure pip is available on the system. Also not sure if the flatpack location above is writable, that might be also a problem with flatpack...

furgo16, 2024-06-25:
> So basically our problem is mostly linux distributions 

Right, and specifically this issue is about sandboxesd packages (snap, flatpak). For those, the system's pip is not accessible. The snap package addresses this by providing its own freecad.pip command. With it, PyPi packages can then be installed.

This worked well  from the command line until recent versions of IfcOpenShell. However, the latest IfcOpenShell upstream pip packages fail to install in the sandbox.

There is also the issue that FreeCAD (either via the Add-On Manager or BIM commands) does not call the sandboxed freecad.pip command, but the system's pip instead. Thus installation via the GUI always fails.

My conclusion was that the most effective fix for now was to bundle IfcOpenShell with the packages to avoid user frustration (unless someone steps up who can work through those issues).

yorikvanhavre, 2024-06-26:
Right, we must definitely rely on packaging IfcOpenShell each time and the user does not need to care about it anymore.
Normally the version of IfcOpenShell being shipped now is good and should work throughout the 1.0 cycle.

I'll just make a PR to check for pip and solve the error mentioned in the OP.

PR #14958 (2024-07-02): BIM: Check for pip when running IfcOpenShell update
Fixes #14499
commit 9b390fa7 (2024-07-02): BIM: Check for pip when running IfcOpenShell update - fixes #14499
closed 2024-07-02 by yorikvanhavre