FreeCAD/FreeCAD #14500: Saving Exports by Document

jenkins007, 2024-06-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Import/Export
I work on multiple documents (projects) at the same time. As I go back and forth, I create .stl files and save them back to the directory where the document resides. I have hundreds of projects organized in multiple directory structures to save each document. FreeCAD remembers the last saved location when exporting which is great for working on the same document. But as I go back and forth on open documents, I have to change the file location to where the project file resides. I may do a test print which takes long time, and I will work on a different document, then save an export for that specific document. Saving the .stl file requires changing save place again.  This is repeated over and over. Summary, exporting files the save place defaults to the last saved place.

What would be nice is that saving an export on a specific document, the default location is the same as the document that is being exported. FreeCAD already knows the file structure of the document, so that information is already available. 

I have been using FreeCAD for several years and the development of this software is impressive! I know this is a low priority.  It could be a quick win since the information is available from the parent document. Otherwise, please put it on the desired list. This could be a selection under Import/Export preferences to export to parent document directory (on/off).

### Full version info

```shell
[code]
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile.backup1675008049.406922 1.81.0
  * Curves.backup1675008022.793941 0.4.4
  * A2plus-835445358730bf414fc0628f1bff0bb62a16acc8 0.4.56
  * 3D_Printing_Tools.backup1665457575.744944
  * CurvedShapes.backup1675000334.5132082 1.0.3
  * Manipulator 1.5.5
  * BIM 2021.12.0
  * Lattice2-master 1.0.0
  * 3D_Printing_Tools
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jenkins007, 2024-06-04:
This could also be resolved by having separate instances of FreeCAD open. Although memory issues may be affected should too many instances be opened. I typically have 2-4 documents open almost all the time.

pierreporte, 2024-06-04:
Is this a side effect of #11914?

jenkins007, 2024-06-04:
I don't believe so. if you only work on one document at a time, there is no issue. Only when working on multiples. I've noticed it since version .19. However, https://github.com/FreeCAD/FreeCAD/issues/11914 appears to be very similar as long as each tab knows the file structure, so this could be an explicit requirement for that ticket.

pierreporte, 2025-06-14:
Yes I think that a program options would be a good idea.

closed 2025-07-16 by chennes