FreeCAD/FreeCAD #14508: SpaceMouse behaviour stuck in dominant mode regardless of being enabled or not

BenBergman, 2024-06-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using my SpaceMouse to rotate my model, only the dominant axis of movement is applied. Going into Tools > Customize > Spaceball Motion I have "Dominant mode" unchecked, but I have the same behaviour with it checked.

[image]

Dominant mode is also disabled in the 3Dconnexion software settings.

[image]

The 3Dconnexion demos work correctly.

I tried upgrading 3DxWare to 10.8.4 as #11753 seemed to indicate it might work now, but then I did not get any functionality with my SpaceMouse.

I tried clearing all FreeCAD settings, reinstalling it, reinstalled 3DxWare 10.7.0, added a shortcut button on the SpaceMouse to toggle dominant mode, nothing has had an effect in FreeCAD.

It has been a while since I used FreeCAD on my Mac, so it is possible this was happening before and I somehow didn't notice, but I'm pretty sure this is new behaviour.

### Full version info

```shell
[code]
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-06:
@numo68 @pskowronskiTDx FYI

BenBergman, 2024-06-06:
It just occurred to me that the last time I used my SpaceMouse and FreeCAD on my Mac might have been on macOS 13. If it wasn't still working in the 3Dconnexion demos, I'd be tempted to blame the OS upgrade.

pskowronskiTDx, 2024-06-07:
I doubt this is a new behaviour, as my latest PR with NavLib integration hasn't affected the MacOS platform (the solution remains unchanged there).

BenBergman, 2024-06-28:
In case it helps, I just tried installing the latest dev build of FreeCAD and the SpaceMouse is still behaving as though it is in dominant mode there.

Is there anything else I can provide to help diagnose this issue?

BenBergman, 2024-06-28:
Okay, I think I have some new info!

I happened to try using Bambu Studio and noticed it ALSO was stuck in dominant mode. I have installed 3DxWare 10.8.4 (previously was using 10.7.0) and now Bambu Studio is behaving as expected.

However, FreeCAD 0.21.2 is no longer responding to my SpaceMouse inputs. 0.22.0dev is working though, so I'm guessing the changes from #11753 are not yet in the latest release?

Probably what happened is 3DxWare 10.7.0 happens to *mostly* work with macOS 14, but for some reason gets stuck in dominant mode while it worked fully in macOS 13. Not sure why it isn't stuck in dominant mode in the 3Dconnexion demos though.

pskowronskiTDx, 2024-07-01:
@BenBergman Thank you for reporting the issue. It appears to be related more to the driver than to a particular application. I have forwarded your findings to our QA team.