FreeCAD/FreeCAD #14510: PartDesign : Unexpected fusion of overlapping elements

marcuspollio, 2024-06-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In a single Body with Compound (multi-solid) allowed, overlapping elements produce an unexpected fusion with only some of these overlapping elements :

Here the Cylinders on top of the first Pad are not overlapping :
[image]

Here the Cylinders are overlapping and only 2 of them are fused :
[image]

This situation is not observed with a simpler overlapping case :

[image]

[image]

To reproduce:

1) Open the provided file
2) Select the VarSet called "Parameters" in the Tree View
3) In the Properties View, Data Tab, change the Property called "Cylinder_Diameter" to a value greater than 60mm to make the cylinders overlapping a bit.

### Full version info

```shell
OS: GNU/Linux Ubuntu 22.04 LTS (DE:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37652 (Git)
Build type: Release
Branch: main
Hash: 36a7fd9be42f2b546562fae832446e8420f9db0c
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk , OCC 7.6.3
```
```

### Subproject(s) affected?

PartDesign

### Anything else?

[Overlap1.zip](https://github.com/user-attachments/files/15571274/Overlap1.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

marcuspollio, 2024-06-07:
> Does this only happen when using VarSets or is this a general Part Design problem?

Hi @maxwxyz, no it does not seam related to VarSets as it's easily reproducible without (it was just easier in this example case to show the issue quickly without opening the sketch or properties).

How can I tell if it's related to TNP or to the Sketcher solver ?

bgbsww, 2024-06-07:
Know?  No.  But playing with it ...  A quick test with a sketch with four circles spits out a bunch of ElementMap errors, I think it is a TNP problem.  So we'll add it to the list ...

marcuspollio, 2024-06-07:
> I think it is a TNP problem. So we'll add it to the list ...

@bgbsww Thank you for having a look ! Let me know if there's anything I can do.

bgbsww, 2024-06-07:
Actually, it looks like a new feature request.  If you trim the overlapping edges so there is just an outline it works fine.  So we're really talking about a new capability here.  This does not appear to work correctly and consistently in either LS3 or 0.21.2.

I think this would require an autoremove of internal edges that do not represent holes in the sketch at the time we create a face from it - a better face generation method.  But it looks like a new capability to me; specific to sketch and not to part design.

marcuspollio, 2024-06-08:
@bgbsww 
I totally agree and I'm OK with not resolving this issue for the time being. This kind of sketch makes no sense in the first place.

IMHO, it's more a matter of User eXperience along the lines of [Prevent to build fragile models](https://freecad.github.io/DevelopersHandbook/roadmap/#objective-prevent-shooting-yourself-in-the-foot). To avoid trying to solve problems caused by flimsy models sounds sensible.
In this particular case, when overlapping geometry is detected, we could imagine a popup with:
"Sketch01 Warning:
Circle01, Circle02, Circle03 and Circle04 are overlapping. Overlapping geometry is not supported.
Button 1: Trim automatically (overlapping geometry to only keep valid geometry and convert the rest to construction geometry)
Button 2: Open Sketch (to solve the issue by yourself)
Button 3: Cancel previous operation (adding overlapping geometry or changing a constrain that produce overlap)
Button 4: Ignore warning"
Something for later in any case.

pieterhijma, 2024-06-08:
> > Does this only happen when using VarSets or is this a general Part Design problem?
> 
> Hi @maxwxyz, no it does not seam related to VarSets as it's easily reproducible without (it was just easier in this example case to show the issue quickly without opening the sketch or properties).

@marcuspollio, thanks for checking and to @maxwxyz, indeed, this is not related to VarSets, at least it would have highly surprised me :slightly_smiling_face:.

Vincenzoferrara, 2024-06-16:
I think it's a tool like cut, but instead of cutting it sets it as internal geometry

marcuspollio, 2024-08-23:
Hi @maxwxyz
Could you please set the milestone to post-1.0, as it needs some discussion and won't be solved for 1.0
Thanks !