FreeCAD/FreeCAD #14511: Reverse the order of entering expressions using the mouse.

kaiwas, 2024-06-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The current order of expressions allows you to take an object parameter using keyboard input.
Below we will look at an example for inserting windows.

Current order.
1) Remember the name of the wall and the name of the vertex.
2) Start typing "Wall."
3) Then "Shape."
4) Then "Vertex No."
5) Then the coordinate. For example "X".

Possible new option.
1) Enter ".X"
2) Click on the desired vertex.
3) The data in the expression will be automatically generated and take the form "Wall.Shape.Vertex11.X"

https://github.com/FreeCAD/FreeCAD/assets/32273833/dba68b66-1e0b-4ca9-a40c-6c77f7dd7238

This proposal is a concept and parameters may vary.
This will allow you to enter only the very “ending” expression from the keyboard. Everything else (the beginning) is calculated automatically, calculated when you click the mouse and depends on the entered “tail” of the expression.

### Full version info

```shell
any
```

### Subproject(s) affected?

Expressions

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-06-05:
Why do you want to enter the tail first?

kaiwas, 2024-06-05:
Yes. The discussion in the link is somewhat similar to this.
Starting input from the end can be convenient if we know the desired parameter in advance.
What is suggested in the link would also solve this problem (just from the other side))))

These are two different approaches, but based on taking information when clicking the mouse. Each has its own advantages.
In this case, we are limited to the parameters.
In the case of a link, we get the name, but after that we will need to enter the same “tail”. And sometimes he is different.
For example, there is: “Wall.Shape.Volume”, and there is “Wall.Shape.Vertex2.X”

If we write ".X" - all that long phrase is inserted automatically and there is no need to write everything that is between the name and the last operator.
That is, with this approach we forcibly limit the sample.

kaiwas, 2024-06-05:
When the “tail” can be used in several places at once, it is possible to draw a conclusion of all possible options that can be selected with the mouse.
It doesn't have to be just vertices or edges. This can be the object itself as a whole (if its parameters have a “tail” entered.)
[image]

maxwxyz, 2024-06-06:
I guess if this is a necessary feature it should be implemented in https://github.com/FreeCAD/FreeCAD/issues/12146

closed 2024-06-06 by maxwxyz