FreeCAD/FreeCAD #14513: New CAM Simulator, not possible to dislay only selected operations.

jkvfc, 2024-06-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Regardless of which operations are selected, all operations are always displayed.
[image]

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37649 (Git)
Build type: Unknown
Branch: main
Hash: 681e8c9d2f95d4341e48cec629267182b2f7e586
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * MyScripts
```

### Subproject(s) affected?

CAM/Path

### Anything else?

[TestPlateManufacturing.zip](https://github.com/user-attachments/files/15576144/TestPlateManufacturing.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

shaise, 2024-06-05:
Bug confirmed. working on it.

shaise, 2024-06-05:
Thanks. A fix PR is pending

closed 2024-06-16 by sliptonic