FreeCAD/FreeCAD #14516: PartDesign_MoveFeature does not properly cleanup OutList of old Body

Roy-043, 2024-06-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create a Body.
2. Add a PartDesign_AdditiveBox.
3. Check the OutList: `obj.OutList => [<PartDesign::AdditiveBox>, <App::Origin object>, <PartDesign::AdditiveBox>]`
4. Create another Body.
5. PartDesign_MoveFeature the feature to the 2nd Body.
6. Result: 'out of the allowed scope' warning.
7. Again check the OutList of original Body: `obj.OutList => [<PartDesign::AdditiveBox>, <App::Origin object>]`.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37645 (Git)
Build type: Release
Branch: main
Hash: fc714933a7e2423d5fa4db640af549814061c42e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2025-07-02:
Should be prioritized since it prints esoteric error to the terminal and confuses users.

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/a050ecce158e4847d0eea81b63e04c8d64c5ba64

It clears the old Body's tip when moving its last features out, so that link doesn't keep the moved feature in its OutList.