FreeCAD/FreeCAD #14520: PDF import. Incorect backgriond color

Kuzma30, 2024-06-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From Import menu
[image]
From "Tools..->Load image" 
[image]

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37476 (Git)
Build type: Release
Branch: step_codepage
Hash: 5c267456625ec98b10a8939750536eef52ed9c4c
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1.dev
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * sheetmetal 0.4.14
  * fasteners 0.5.20
  * addFC 0.2.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Kuzma30, 2024-06-05:
Test file
[B32-X.PDF](https://github.com/user-attachments/files/15593279/B32-X.PDF)

wwmayer, 2024-06-06:
Currently PDF files are loaded as image and it looks like a Qt problem. For the future we can use Qt's PDF library that allows to load a PDF file as real document.

Kuzma30, 2024-06-06:
> Currently PDF files are loaded as image and it looks like a Qt problem. For the future we can use Qt's PDF library that allows to load a PDF file as real document.

Tools -Load image used Qt way (QImage) and it works normally. "Import" used STD_Import way and I don't found how this realised.

tetektoza, 2025-04-13:
Is loading PDF still supported from this menu? I see that @Kuzma30 called `Std_ViewLoadPDF` on the second screenshot on python console, although I don't see this command ANYWHERE in the code. And it also seems I can't import PDF anymore from under `Tools->Load Image`.

FEA-eng, 2025-11-14:
See my comment [here](https://github.com/FreeCAD/FreeCAD/issues/14520#issuecomment-3289711804)

Kuzma30, 2026-08-27:
Make test on main branch
File - Open.. (dark theme)
[image]
File - Open.. (light theme)

[image]

OS: Ubuntu 26.04 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 26.3.0dev.48253 (Git)
Build date: 2026/08/24 23:19:15
Build type: Release
Branch: main
Hash: b352a4fc7f27dbf5464ab48685838254f3f05869
Python 3.14.4, Qt 6.10.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.5.2, boost 1_90, Eigen3 3.4.0, PySide 6.10.2
shiboken 6.10.2, SMESH 7.7.1.0, xerces-c 3.2.4, Clipper2 , OCC 7.9.2
Locale: Ukrainian/Ukraine (uk_UA)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
QPA/File dialog/Color dialog: wayland/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: Gesture/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/101.6/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 580.173.02/NVIDIA Corporation/NVIDIA GeForce GTX 1060 6GB/PCIe/SSE2
Installed mods:

On light theme it usable. I have proposal - make option to enable/disable transparent.

[0001-fix-14520-pdf-color-inversion(1).patch](https://github.com/user-attachments/files/31505048/0001-fix-14520-pdf-color-inversion.1.patch)

Kuzma30, 2026-08-27:
With proposed patch
Activated option
[image]
Disabled option

[image]

Kuzma30, 2026-08-30:
### Native PDF viewer with text selection, search and vector export

Built a `Gui::PdfView` — an MDIView wrapping `QPdfView` from Qt PDF — that opens PDFs as live documents instead of rasterising them into an ImagePlane.

**What it does:**

- **Text search** — Ctrl+F opens a search bar backed by `QPdfSearchModel`. Matches are highlighted by `QPdfView`, with next/previous navigation and a match counter.
- **Text selection** — `PdfSelectionView` (a `QPdfView` subclass) adds mouse selection via `QPdfDocument::getSelection()`. Ctrl+C copies the selection, Ctrl+A selects the whole page. Context menu has "Copy page text" / "Copy all text" via `getAllText()`.
- **Region crop → ImagePlane** — toggle the "Region" button, drag a rectangle on the page, right-click → "Insert region as image plane". Renders the crop at 300 DPI using `QPdfDocumentRenderOptions` (`setScaledSize` + `setScaledClipRect`), saves to PNG, creates an `Image::ImagePlane` in the document tree.
- **Region crop → Sketch (vector)** — right-click → "Import region as sketch (vector)". Calls `pdftocairo -svg` with `-x -y -W -H` to extract the region as an SVG with vector paths, then `Draft.importSVG` brings lines/arcs/curves into the document. Requires `poppler-utils` at runtime.
- **Zoom** — in/out and fit-to-width.
- **Printing** — through the standard MDIView print/export mechanism.

**Build:** optional, needs `Qt6::Pdf` + `Qt6::PdfWidgets` ≥ 6.4. Without them FreeCAD compiles and runs exactly as before — PDF falls back to the existing `Image::ImagePlane` raster path. `poppler-utils` is an optional runtime dependency for vector import only.

**Routing:** `FileHandler` intercepts `.pdf` before `QImageReader` and opens it in `PdfView` when `HAVE_QT_PDF_WIDGETS` is defined.

[0004-native-pdf-view-qtpdf.patch](https://github.com/user-attachments/files/31611685/0004-native-pdf-view-qtpdf.patch)

https://github.com/user-attachments/assets/aab6dc55-674f-4648-9871-3e07a9c44109

Kuzma30, 2026-09-05:
https://github.com/user-attachments/assets/a8e6f844-b737-421f-98ee-20fa2c2f4849
https://github.com/user-attachments/assets/b0fef1bd-336e-499a-a2af-6ec6287e0233
https://github.com/user-attachments/assets/e9d012ac-0210-43dd-8e9e-15848c597ec5

https://github.com/Kuzma30/FreeCAD/tree/documentation_wb still have some bugs but in general its working. You can integrate documentation to FreeCAD file, native open pdf in program with ability select region and use it in FreeCAD as ImagePlane. If community will be interested in such changes I will continue work in this direction.

m1cha1, 2026-09-05:
Nice but as I remeber  @wwmayer  has something similar  https://github.com/wwmayer/FreeCAD/tree/qt_pdf

Kuzma30, 2026-09-06:
Don't see this branch before. Take some ideas from @wwmayer and upgrade my branch.