FreeCAD/FreeCAD #14521: Edge selection not shown in CAM model

Anga23, 2024-06-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I try to select multiple edges only the first one is highlighted.
You can reproduce it also if you use "CAM_SelectLoop".
This happend only in the CAM workbench on the model, in the part design workbench everything is working fine.

### Full version info

```shell
OS: OS 22.04 LTS (GNOME)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git) AppImage
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods:

fasteners 0.4.70
FeedsAndSpeeds 0.5.0
MakerWorkbench 1.0.1
sheetmetal 0.3.1
```

### Subproject(s) affected?

CAM/Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-06-05:
Confirmed, educated guess based on feedback from Sheetmetal Wb, is that https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/CAM/Path/Op/Gui/Selection.py needs updating to be TNP output aware, this could affect quite a few addons IMHO.

bgbsww, 2024-06-06:
Selection.py is exactly the same in LS3, and it works there.

Since the selection box clearly shows the selection is happening, this is a display issue, which I think angles towards the ViewProviders and not the python, no?  What's the feedback from Sheetmetal WB?

Syres916, 2024-06-06:
@bgbsww the fixes that have been made to Sheetmetal Wb are https://github.com/shaise/FreeCAD_SheetMetal/commit/9b8e2cd18653c2f3017fbf66e3e183fd41db7112 and https://github.com/shaise/FreeCAD_SheetMetal/commit/b070c508bfe015eedaded13b3b25a88e66972f04
@shaise mentioned that realthunder's branch alongside FreeCAD 0.21.x and 0.22dev worked fine with Sheetmetal Wb (therefore for many months) prior to these very recent changes made to work with 0.22dev with TNP activated. To be certain I'm running a _git bisect_ and will report my findings, I'm currently down to 90 commits but sleep beckons. The CAM issue only gets triggered once a job is created for the geometry in question, it's not just activating the workbench.

bgbsww, 2024-06-06:
Thanks!  I can chase from there.

PR #14699 (2024-06-17): Toponaming: fix secondElementSelection and SelectionView
This fixes #14521 the reported bug first seen in selections of CAM model objects where only one selection could be made.

It also cleans up the Selection View to show short element names, and not full history names.

Fix #13248
closed 2024-06-17 by chennes