FreeCAD/FreeCAD #14527: [Assembly] : Limitations of the FreeCAD Slider

luisDias89, 2024-06-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@PaddleStroke  Continuation of the topic https://github.com/FreeCAD/FreeCAD/issues/14488

Limitations 1: 
  https://github.com/FreeCAD/FreeCAD/assets/48370099/99a2ed1c-883b-42c7-8cb6-67c1d65baad8
Limitations  2:     (This is the one I've already described in the topic I mentioned)
  Slider with gap:
  [image]

The FreeCAD slider tool (similar to the one in OneShape) only solves simple slider problems. When dealing with more complex mechanisms, it's necessary to resort to other, more primitive tools. I'll show a video of how to solve the problem in SolidWorks in the simplest way possible:

https://github.com/FreeCAD/FreeCAD/assets/48370099/2c759cb2-b302-433f-bbd0-879d8d0ef5e9

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37645 (Git)
Build type: Release
Branch: main
Hash: fc714933a7e2423d5fa4db640af549814061c42e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Portugal (pt_PT)
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

qewer33, 2024-06-06:
The firs issue isn't an issue with the Joint, you're just selecting the wrong elements for the JCSs:

[vokoscreenNG-2024-06-06_14-11-30.webm](https://github.com/FreeCAD/FreeCAD/assets/69015181/63a3a262-6642-4b2c-952a-dcf1842e6b55)

luisDias89, 2024-06-06:
@qewer33 Yes, you are right. It's possible. Now it just needs to handle the GAP in all cases.

PaddleStroke, 2024-06-06:
For limitation 2: I think the best would be to add the possibility to set offset along the X and Y axis.

howie-j, 2024-06-06:
@luisDias89 As i said previously, the new assembly workbench works very differently to Solidworks, and much more similar to Onshape. The main differences is that Solidworks uses simple constraints on single elements (faces/curves/vertexes) and the Assembly workbench uses joints with JCS (Joint Coordinate System, that preview 3 axis widget you see when selecting elements). That means for a fixed joint, all 3 rotational axis and all 3 translate axis are locked. For a slider, all axis except for Z axis translate is locked. The orientation of the JCS when selecting something is critical for making the joint work correctly. The limitation 1 video is an example of such a incorrectly made joint, as @qewer33 showed the correct way to make that joint.

This also means only one joint is almost always enough per part. No need for 3 constraints (2x coincident + distance limit) as in Solidworks, a single slider joint does the same. This is a MUCH better system in most cases, as it reduces the total number of joints/constraints, simplifies the assembly greatly and makes it so much easier to maintain and fix errors in the assembly later.

To make more advanced joints such as the example with a gap, you can select either sketch elements or datum elements in the original parts as joint elements. Here i simply created two datum lines along the common axis for both sliders: [Slider_gap.zip](https://github.com/user-attachments/files/15657408/Slider_gap.zip)  

[slider_gap.webm](https://github.com/FreeCAD/FreeCAD/assets/82042580/456a0226-3768-4516-9245-c88a2ed87110)

Onshape solves this issue multiple ways:

1. By Mate Connectors, which are standalone JCS objects you can place in bodies or parts. FreeCAD does not have this yet, but it might come in the next release after 1.0. In the meantime, using datum points, lines, local coordinate systems and sketches with sketch elements works great.
2. By modifying the JCS orientation and offset upon joint creation:  
[image]  
This is similar to what @PaddleStroke proposes i guess, and is essential for more advanced assembly work. This will not be in 1.0, but should absolutely be included in the next release. I might try implementing it myself if i find time.

Remember that Assembly workbench is super new and fresh, and under heavy development. The goal for 1.0 was to have a working MVP, not an Onshape/Inventor/Solidworks Assembly replacement. If you still struggle with the idea of joints instead of constraints, i recommend switching to the addon A2plus, which works much more similar to Solidworks.

closed 2024-06-06 by maxwxyz

howie-j, 2024-06-06:
Well i'll be damned, joint offset already exist, it's just not exposed in the joint task panel yet. So no auxillary/datum geometric elements are actually needed:  

[offset.webm](https://github.com/FreeCAD/FreeCAD/assets/82042580/dd67ee5b-b954-4e1e-9a8f-f3e4813aaf63)

luisDias89, 2024-06-06:
@howie-j Yes, that's it, but in an intuitive way with the Slider

PaddleStroke, 2024-06-07:
The UI just needs to add spinboxes for X Y offset when they are needed