FreeCAD/FreeCAD #14530: [Part] Make selected solid transparent to enable selecting hidden elements

PaddleStroke, 2024-06-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From the forum : https://forum.freecad.org/viewtopic.php?t=88214

Here's a clip of what I'm talking about. In LinkStage, when an object is selected it becomes transparent and obscured faces/lines/points can be selected easily.

https://github.com/FreeCAD/FreeCAD/assets/19984177/1b8e7c82-29af-4b7d-b7d7-b516c97fcc99

@realthunder is this feature possible to port easily?

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-06-06:
Is it really the best solution? It works well for standalone parts but not in assembly context. Even if the selected part was visible like proposed in #13981, the part needs to be visible to be selected and parts buried into an assembly are common.

In Catia, this problem has been solved with a local x-ray mode:

[image]

When hitting up or down arrow, a circle appears around the cursor. It is at a fixed size determined by the position of the mouse at the tool activation. A list appears near the circle with the location of the element. Its content is every geometry element (surfaces, edges, etc.) below the circle center and inside picking radius. Hitting up or down circles in this list. The current selection is displayed as opaque while other are transparent. This feature also allows to select coincident geometry.

More info: https://www.technia.com/blog/help-with-selecting-elements-in-catia-v5/

Other cad software have similar features, like Solid Edge.

maxwxyz, 2024-06-06:
Is this issue just about adding transparency or is it about better selection?
Then I guess it would be related to https://github.com/FreeCAD/FreeCAD/issues/10308#issuecomment-1783420363

eblanshey, 2025-04-03:
Solid Edge's equivalent feature is called "Quick Pick". Their solution is to place the mouse over any element in the 3D view and wait 2 seconds (configurable). After the mouse stops moving, a little box pops up that contains all elements under the cursor with some margin, including elements hidden behind other elements. As you mouse over each element in the box, the element is highlighted in the 3d view. [Link to relevant docs.](https://www.soliddna.com/SEHelp/ST5/EN/i_v/selectb1h.htm) If used in combination with a visible circle + keyboard arrow selections like @pierreporte mentioned it'd be very powerful.

It's worth noting that the same feature would be very useful in sketching mode, when multiple lines, points, or constraints overlap (e.g. multiple coincident constraints.)