FreeCAD/FreeCAD #14531: Update OCT version on daily PPA to 7.7.2

furgo16, 2024-06-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Debian package in the [FreeCAD daily PPA](https://launchpad.net/~freecad-maintainers/+archive/ubuntu/freecad-daily) is behind in the OCT version it ships (7.6.3) when compared to the Linux AppImage (7.7.2).

It would be good to bring them inline to avoid reporting issues that might have been fixed in later OCT versions.

This is controlled by the build recipe at https://code.launchpad.net/~freecad-maintainers/+recipe/opencascade-ppa. There is no way to submit changes for this on Launchpad, but essentially this small diff on the recipe would make the build use 7.7.2:

```diff
diff --git a/recipe b/recipe
--- a/recipe
+++ b/recipe
@@ -1,3 +1,3 @@
 # git-build-recipe format 0.4 deb-version 1:{debupstream}-8
-lp:~freecad-maintainers/opencascade/+git/main ppa/master
+lp:~freecad-maintainers/opencascade/+git/main upstream
```

Another option, maybe better, would be to merge the `upstream` branch into `ppa/master`

Note: I did not find a repo in GitHub for the PPA packaging, thus submitting the issue here.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37569 (Git)
Build type: Release
Branch: main
Hash: dd7a8c3eba49e160f870536ec20af1a069a3be1e
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Installed mods: 
  * OpenTheme 2024.5.3
  * parts_library
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-06-06:
Surely if you want to make a change as significant as that you should advise the pros and cons. Is this https://github.com/FreeCAD/FreeCAD/issues/9651 something that's irrelevant? I would never use a version of OCCT for real world work that had such a known issue but that's my personal opinion, I use 7.3.3 occasionally but predominantly 7.6.3.

FlachyJoe, 2024-06-06:
FYI debian sid was upgraded from 7.6.3 to 7.8.1 directly

furgo16, 2024-06-06:
My reasoning was:

1. The weekly AppImage ships with OCT 7.7.2, so weighing the pros and cons that lead to that decision had already happened
2. AppImage being one of the official installation methods, that's the OCT version that will be shipped in the 1.0 release
3. As such, it might make sense to update the other daily packages (.snap, .deb) to use the same OCT version
4. Also, given that we're past feature freeze, an upgrade to OCT 7.8.1, which allegedly fixes #9651, will not happen for the 1.0 release

I do not claim that reasoning to be correct, I'm simply a contributor attempting to improve FreeCAD with my limited skills and time. I was also probably wrong on at least point 2 => building FreeCAD from source on Linux results on a binary that uses OCT 7.6.3.

In any case, I think harmonizing the OCT versions on the different weekly packages (7.6.3, 7.7.1, 7.7.2) might be worth considering. At the very least to avoid confusion and increased issue triage time. Regardless of whether that means a version upgrade or a downgrade.

Syres916, 2024-06-11:
I'm going to do some end-to-end testing with OCCT 7.8.1 (purposely using TNP avoidance modelling techniques) but the first thing I hit was only being able to compile with _ENABLE_DEVELOPER_TESTS=Off_ as I got an error:

```
/home/username/freecad-main-build781/tests/Part_tests_run: error while loading shared libraries: libTKDE.so.7.8: cannot open shared object file: No such file or directory
CMake Error at /usr/share/cmake-3.16/Modules/GoogleTestAddTests.cmake:40 (message):
  Error running test executable.

    Path: '/home/username/freecad-main-build781/tests/Part_tests_run'
    Result: 127
    Output:
      

make[2]: *** [tests/CMakeFiles/Part_tests_run.dir/build.make:543: tests/Part_tests_run] Error 1
make[2]: *** Deleting file 'tests/Part_tests_run'
make[1]: *** [CMakeFiles/Makefile2:9435: tests/CMakeFiles/Part_tests_run.dir/all] Error 2
```
trying to compile as I normally would.

I'll report back with my findings later this week, if need be I'll prepare Unit Tests for any reproducible faults I may find.

Syres916, 2024-06-14:
My update isn't very thorough as most of testing has just flushed out 0.22dev issues but there is one thing that 7.8.1 worked where 7.6.3 didn't but it was quite obscure in TechDraw and there's no way it would have featured in my day-to-day work see https://forum.freecad.org/viewtopic.php?t=88440
I've got quite a bit of home DIY this weekend so it'll be Monday for the next update.

Syres916, 2024-06-19:
Sorry for the delay, I've been successful with some complex surfacing work and while dealing with PR's breaking compiling. The last thing I need to do is bombard 7.8.1 with some of my python scripts that build sketches and features. My confidence compared to 7.7.2 is certainly higher with 7.8.1.

Syres916, 2024-07-06:
I have a model with over 400 faces which is giving trouble when they are all being interated for CfdOF analysis so I've not much choice but to recompile with TNP disabled in order to ensure where the fault is originating from. The same model in 0.21.2 with OCC 7.6.3 is rock stable.

adrianinsaval, 2024-07-14:
I have updated the daily package for 22.04 to ooct 7.8.1 now

adrianinsaval, 2024-07-14:
> I have a model with over 400 faces which is giving trouble when they are all being interated for CfdOF analysis so I've not much choice but to recompile with TNP disabled in order to ensure where the fault is originating from. The same model in 0.21.2 with OCC 7.6.3 is rock stable.

if this is still a problem please report it as it's own issue, this might not have much to do with occt version if the difference is between enabling or disabling tnp mitigation

adrianinsaval, 2024-07-21:
updating 20.04 to 7.8.1 is not worth doing since it requires updating other dependencies too and we'll likely drop support next year anyways. I'll close this issue since it's been completed were reasonable

closed 2024-07-21 by adrianinsaval