FreeCAD/FreeCAD #14532: Update file thumbnail logo to new FreeCAD logo

furgo16, 2024-06-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently saving a document with option "Preferences > General > Document > Add the program logo to the generated thumbnail" adds the old FreeCAD logo, where it should be adding the new FreeCAD logo instead.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37578 +9 (Git)
Build type: Unknown
Branch: Draft_Grid_Observer_clean
Hash: 01b2b370e9b7538eb20ab07dd20ff036e36f51e5
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Installed mods: 
  * OpenTheme 2024.5.3
  * parts_library
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-06-06:
This has already been fixed.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37645 (Git)
Build type: Release
Branch: main
Hash: fc714933a7e2423d5fa4db640af549814061c42e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

closed 2024-06-06 by Roy-043

furgo16, 2024-06-06:
Indeed it has, I was looking at an older build, apologies.

[image]