FreeCAD/FreeCAD #14533: Update examples files thumbnail logo to new FreeCAD logo

furgo16, 2024-06-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If and when issue #14532 is fixed, it would be worth re-saving the example files shipped in the examples folder to use the new FreeCAD logo, all in the same size.

[image]

As an additional consideration, I'd argue that the logo should be removed from the example files. It does not add any particular value to the examples displayed within FreeCAD, and it rather obscures the thumbnail. But others might not agree, so it might be better to discuss it separately.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37578 +9 (Git)
Build type: Unknown
Branch: Draft_Grid_Observer_clean
Hash: 01b2b370e9b7538eb20ab07dd20ff036e36f51e5
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Installed mods: 
  * OpenTheme 2024.5.3
  * parts_library
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-06:
I'd propose to get rid of the thumbnail icon file, because at least on windows it is rendered on top of the image in the file explorer so it's twice there.

pierreporte, 2024-06-06:
I’m in favor of removing the icon from the preview. However, I think that an indication of the file type is still useful. The screenshot shows a step file in the tree. If a preview is eventually implemented (same for STLs and other formats), some sort of marker could help distinguish file types. It could be an icon in the label.

furgo16, 2024-06-27:
Related forum thread: https://forum.freecad.org/viewtopic.php?t=81005

luzpaz, 2024-07-24:
Should this be in 1.0 ?

maxwxyz, 2024-07-24:
If possible, the logo integration in the thumbnail would be removed.

luzpaz, 2024-08-03:
Hot take: this should be a UI blocker because it is a the first thing a user sees when firing up FreeCAD. So we either add the new logo or remove the logo as an overlay for the thumbnail.

maxwxyz, 2024-08-04:
My proposal is disable the icon implementation by default (can be enabled in the preferences) and save all example files without the embedded logo (as on Windows it's rendered on top and then appears twice).

closed 2024-08-12 by chennes