FreeCAD/FreeCAD #14534: Measurement: circle length

boboxx, 2024-06-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

using the new measurement tools FreeCAD crashes if you try to measure a circle length
[test-length.FCStd.zip](https://github.com/user-attachments/files/15614561/test-length.FCStd.zip)

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37645 (Git)
Build type: Release
Branch: main
Hash: fc714933a7e2423d5fa4db640af549814061c42e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * OpenTheme 2024.5.16
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-06-06:
For me, there's no crash. It just fails:

[image]

and shows this error message in the report view:

`0 Selection.cpp(159): Unhandled unknown exception caught in selection observer`

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37645 (Git)
Build type: Release
Branch: main
Hash: fc714933a7e2423d5fa4db640af549814061c42e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

maxwxyz, 2024-06-06:
same as FEA-eng on Windows.
@hlorus FYI

FEA-eng, 2024-06-06:
The effect is the same as in #14411 (0 mm label away from the measured entity and "Unhandled unknown exception caught in selection observer" error message). Apparently, that's what happens when trying to do unsupported measurements. It should be changed to something more user-friendly.

PR #14591 (2024-06-17): Measure: Fix MeasureLength failure when measuring circles/arcs
Currently MeasureLength fails when trying to measure circles and arcs. The relevant code where the exception occurs can actually be be simplified as the rotation of the measured element is not needed to place the measure label.

Fixes #14411 
Fixes #14534
closed 2024-06-17 by WandererFan