FreeCAD/FreeCAD #14535: FreeCAD crashes upon window tool use from Arch workbench

gnthibault, 2024-06-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Whenever I try to add a window on a wall with Arch workbench, FreeCAD consistently crashes. I cannot continue my project.
[freecad_design.FCStd.zip](https://github.com/user-attachments/files/15621340/freecad_design.FCStd.zip)

This should be reproducible by simply adding a window on a wall (doesn't matter which one). I usually enable "Snap near" but with or without this, I got a crash.

Any workaround welcome

Thank you in advance for your help

### Full version info

```shell
[code]
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-06-06:
No crash here. Please give more details.
[image]
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37645 (Git)
Build type: Release
Branch: main
Hash: fc714933a7e2423d5fa4db640af549814061c42e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

kaiwas, 2024-06-06:
> Thank you in advance for your help

Try clearing your cache.
I think I had this problem once.

Is the problem reproducible in the new file or does it only appear in the current project?

xtemp09, 2024-06-06:
> OS: macOS 14.2.1

This is the problem.

@gnthibault, could you provide the backtrace?

gnthibault, 2024-06-06:
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

(qt.qpa.fonts) Populating font family aliases took 166 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 
[58664:19715:0606/223230.244190:ERROR:command_buffer_proxy_impl.cc(141)] ContextResult::kTransientFailure: Failed to send GpuChannelMsg_CreateCommandBuffer.
Fontconfig error: Cannot load default config file: No such file: (null)
Choose a face on an existing object or select a preset
/Applications/FreeCAD.app/Contents/MacOS/FreeCAD: line 19: 58662 Segmentation fault: 11  "${PREFIX}/bin/freecad" $@
logout

Saving session...completed.
Deleting expired sessions...1 completed.

[Process completed]

gnthibault, 2024-06-06:
Here is the report
[report.txt](https://github.com/user-attachments/files/15686242/report.txt)

xtemp09, 2024-06-07:
The backtrace provided is really-really close to #13990 and #13915.

xtemp09, 2024-06-07:
@gnthibault, can you reproduce the crash with the latest development version? _BIM_ workbench has undergone changes recently, connected to the backtrace provided.

gnthibault, 2024-06-07:
Hmm Installed the last dev release, but it looks I am still missing something, sorry:

> 21:29:11  (qt.qpa.fonts) Populating font family aliases took 173 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 
> 21:29:12  Migrating Start Workbench to Start command... done.
> 21:29:15  BIM addon path has been renamed to BIM021 to avoid conflicts with the builtin BIM workbench. Please restart FreeCAD
> 21:29:45  pyException: Traceback (most recent call last):
>   File "/private/var/folders/17/pnktwcq50hgcnjjs3f3khg980000gn/T/AppTranslocation/148B51FA-05D7-42BA-9396-14916A6204D2/d/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
>     return original_import(name, *args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> <class 'ModuleNotFoundError'>: No module named 'ArchStructure'
> 21:29:45  pyException: Traceback (most recent call last):
>   File "/private/var/folders/17/pnktwcq50hgcnjjs3f3khg980000gn/T/AppTranslocation/148B51FA-05D7-42BA-9396-14916A6204D2/d/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
>     return original_import(name, *args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> <class 'ModuleNotFoundError'>: No module named 'ArchWall'
> 21:29:45  pyException: Traceback (most recent call last):
>   File "/private/var/folders/17/pnktwcq50hgcnjjs3f3khg980000gn/T/AppTranslocation/148B51FA-05D7-42BA-9396-14916A6204D2/d/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
>     return original_import(name, *args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> <class 'ModuleNotFoundError'>: No module named 'ArchWindow'
> 21:29:45  pyException: Traceback (most recent call last):
>   File "/private/var/folders/17/pnktwcq50hgcnjjs3f3khg980000gn/T/AppTranslocation/148B51FA-05D7-42BA-9396-14916A6204D2/d/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
>     return original_import(name, *args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> <class 'ModuleNotFoundError'>: No module named 'ArchWindow'
> 21:29:45  pyException: Traceback (most recent call last):
>   File "/private/var/folders/17/pnktwcq50hgcnjjs3f3khg980000gn/T/AppTranslocation/148B51FA-05D7-42BA-9396-14916A6204D2/d/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
>     return original_import(name, *args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> <class 'ModuleNotFoundError'>: No module named 'ArchWindow'
> 21:29:45  pyException: Traceback (most recent call last):
>   File "/private/var/folders/17/pnktwcq50hgcnjjs3f3khg980000gn/T/AppTranslocation/148B51FA-05D7-42BA-9396-14916A6204D2/d/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
>     return original_import(name, *args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> <class 'ModuleNotFoundError'>: No module named 'ArchWindow'
> 21:29:45  pyException: Traceback (most recent call last):
>   File "/private/var/folders/17/pnktwcq50hgcnjjs3f3khg980000gn/T/AppTranslocation/148B51FA-05D7-42BA-9396-14916A6204D2/d/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
>     return original_import(name, *args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> <class 'ModuleNotFoundError'>: No module named 'ArchWindow'
> 21:29:45  pyException: Traceback (most recent call last):
>   File "/private/var/folders/17/pnktwcq50hgcnjjs3f3khg980000gn/T/AppTranslocation/148B51FA-05D7-42BA-9396-14916A6204D2/d/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
>     return original_import(name, *args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> <class 'ModuleNotFoundError'>: No module named 'ArchStructure'
> 21:29:45  pyException: Traceback (most recent call last):
>   File "/private/var/folders/17/pnktwcq50hgcnjjs3f3khg980000gn/T/AppTranslocation/148B51FA-05D7-42BA-9396-14916A6204D2/d/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
>     return original_import(name, *args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> <class 'ModuleNotFoundError'>: No module named 'ArchWindow'
> 21:29:45  pyException: Traceback (most recent call last):
>   File "/private/var/folders/17/pnktwcq50hgcnjjs3f3khg980000gn/T/AppTranslocation/148B51FA-05D7-42BA-9396-14916A6204D2/d/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
>     return original_import(name, *args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> <class 'ModuleNotFoundError'>: No module named 'ArchWall'
> 21:30:10  No module named 'Arch_rc'
> 21:30:10  Traceback (most recent call last):
>   File "/Users/gnthibault/Library/Application Support/FreeCAD/Mod/BIM/./InitGui.py", line 77, in Initialize
[... cut ...] ure.py", line 139, in _import
>     return original_import(name, *args, **kwargs)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/private/var/folders/17/pnktwcq50hgcnjjs3f3khg980000gn/T/AppTranslocation/148B51FA-05D7-42BA-9396-14916A6204D2/d/FreeCAD.app/Contents/Resources/Mod/Draft/draftutils/utils.py", line 42, in <module>
>     from draftutils import params
>   File "/private/var/folders/17/pnktwcq50hgcnjjs3f3khg980000gn/T/AppTranslocation/148B51FA-05D7-42BA-9396-14916A6204D2/
>

xtemp09, 2024-06-08:
> I am not 100% sure this is related though.

Yes, this crash is unrelated.

marcosilveira2, 2024-09-22:
same to me
Choose a face on an existing object or select a preset
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7fdf5b85d090]
#1  0x7fdf57f10239 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x29
#2  0x7fdf580a74d8 in SoChildList::traverse(SoAction*, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x168
#3  0x7fdf580f25ec in SoGroup::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x5c
#4  0x7fdf5810c1ec in SoSeparator::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x2c
#5  0x7fdf5aaa1ce4 in Gui::SoFCUnifiedSelection::handleEvent(SoHandleEventAction*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x94
#6  0x7fdf57f10277 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x67
#7  0x7fdf580a74d8 in SoChildList::traverse(SoAction*, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x168
#8  0x7fdf580f25ec in SoGroup::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x5c
#9  0x7fdf5810c1ec in SoSeparator::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x2c
#10  0x7fdf57f10277 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x67
#11  0x7fdf57f199ab in SoHandleEventAction::beginTraversal(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0xab
#12  0x7fdf57f1071a in SoAction::apply(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x28a
#13  0x7fdf580bf651 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x71
#14  0x7fdf5aa025f2 in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x22
#15  0x7fdf5aa07c0b in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x5b
#16  0x7fdf5a9d388d in Gui::View3DInventorViewer::processSoEventBase(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0xd
#17  0x7fdf5a9fdbbd in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x2d
#18  0x7fdf5a9f6dff in Gui::BlenderNavigationStyle::processSoEvent(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x25f
#19  0x7fdf5a9fe7e6 in Gui::NavigationStyle::processEvent(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x66
#20  0x7fdf5a9d8150 in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x70
#21  0x7fdf5aa1513f in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0xaf
#22  0x7fdf588a651b in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x9b
#23  0x7fdf592bea55 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x75
#24  0x7fdf592c8343 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x583
#25  0x7fdf5accc140 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x90
#26  0x7fdf588a680a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x18a
#27  0x7fdf592c7457 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1b7
#28  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c935d) [0x7fdf5931d35d]
#29  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cc1ec) [0x7fdf593201ec]
#30  0x7fdf592bea66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x86
#31  0x7fdf592c80f0 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x330
#32  0x7fdf5accc140 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x90
#33  0x7fdf588a680a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x18a
#34  0x7fdf58c8f7d3 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x6e3
#35  0x7fdf58c9110b in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x25b
#36  0x7fdf58c6b35b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xbb
#37  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x7932e) [0x7fdf5285b32e]
#38  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7fdf5693217d]
#39  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x52400) [0x7fdf56932400]
#40  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7fdf569324a3]
#41  0x7fdf588fe435 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x65
#42  0x7fdf588a53ab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#43  0x7fdf588ad116 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x96
#44  0x7fdf5ad52770 in Gui::Application::runApplication() from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x1e40
#45  freecad(+0xde54) [0x557828709e54]
#46  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7fdf5b83e083]
#47  freecad(+0xf07e) [0x55782870b07e]

marcosilveira2, 2024-09-22:
During initialization the error ""'BIMWorkbench21' already exists."" occurred in /home/marco/.local/share/FreeCAD/Mod/BIM021/./InitGui.py
Please look into the log file for further information
connect failed: Arquivo ou diretório inexistente
Unknown command 'Draft_Array'
Unknown command 'Draft_Array'
The `.FCStd#` backup format is deprecated as of v0.21 and may be removed in future versions.
To update, check the 'Preferences->General->Document->Use date and FCBak extension' option.
Unknown command 'Draft_Array'
Unknown command 'Draft_Array'
Choose a face on an existing object or select a preset
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7fdf5b85d090]
#1  0x7fdf57f10239 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x29
#2  0x7fdf580a74d8 in SoChildList::traverse(SoAction*, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x168
#3  0x7fdf580f25ec in SoGroup::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x5c
#4  0x7fdf5810c1ec in SoSeparator::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x2c
#5  0x7fdf5aaa1ce4 in Gui::SoFCUnifiedSelection::handleEvent(SoHandleEventAction*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x94
#6  0x7fdf57f10277 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x67
#7  0x7fdf580a74d8 in SoChildList::traverse(SoAction*, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x168
#8  0x7fdf580f25ec in SoGroup::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x5c
#9  0x7fdf5810c1ec in SoSeparator::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x2c
#10  0x7fdf57f10277 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x67
#11  0x7fdf57f199ab in SoHandleEventAction::beginTraversal(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0xab
#12  0x7fdf57f1071a in SoAction::apply(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x28a
#13  0x7fdf580bf651 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x71
#14  0x7fdf5aa025f2 in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x22
#15  0x7fdf5aa07c0b in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x5b
#16  0x7fdf5a9d388d in Gui::View3DInventorViewer::processSoEventBase(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0xd
#17  0x7fdf5a9fdbbd in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x2d
#18  0x7fdf5a9f6dff in Gui::BlenderNavigationStyle::processSoEvent(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x25f
#19  0x7fdf5a9fe7e6 in Gui::NavigationStyle::processEvent(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x66
#20  0x7fdf5a9d8150 in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x70
#21  0x7fdf5aa1513f in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0xaf
#22  0x7fdf588a651b in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x9b
#23  0x7fdf592bea55 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x75
#24  0x7fdf592c8343 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x583
#25  0x7fdf5accc140 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x90
#26  0x7fdf588a680a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x18a
#27  0x7fdf592c7457 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1b7
#28  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c935d) [0x7fdf5931d35d]
#29  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cc1ec) [0x7fdf593201ec]
#30  0x7fdf592bea66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x86
#31  0x7fdf592c80f0 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x330
#32  0x7fdf5accc140 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x90
#33  0x7fdf588a680a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x18a
#34  0x7fdf58c8f7d3 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x6e3
#35  0x7fdf58c9110b in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x25b
#36  0x7fdf58c6b35b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xbb
#37  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x7932e) [0x7fdf5285b32e]
#38  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7fdf5693217d]
#39  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x52400) [0x7fdf56932400]
[... cut ...]
#41  0x7fdf588fe435 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x65
#42  0x7fdf588a53ab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#43  0x7fdf588ad116 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x96
#44  0x7fdf5ad52770 in Gui::Application::runApplication() from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x1e40
#45  freecad(+0xde54) [0x557828709e54]
#46  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7fdf5b83e083]
#47  freecad(+0xf07e) [0x55782870b07e]

marcosilveira2, 2024-09-22:
[code]
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: Portuguese/Brazil (pt_BR)
Installed mods: 
  * BIM.backup1724617972.7890484 2021.12.0
  * BIM 2021.12.0
  * BIM021 2021.12.0
  * OpenTheme 2024.9.1
[/code]

luzpaz, 2024-09-23:
@marcosilveira2 please reproduce on 1.0RC1 https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc1

marcosilveira2, 2024-09-23:
> @marcosilveira2 please reproduce on 1.0RC1 https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/1.0rc1

I tested with 1.0RC1 and BIM Workbench doesn't work.

FreeCAD 1.0.0, Libs: 1.0.0RC1R38642 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

During initialization the error "[Errno 39] Diretório não vazio: '/home/marco/.local/share/FreeCAD/Mod/BIM' -> '/home/marco/.local/share/FreeCAD/Mod/BIM021'" occurred in /tmp/.mount_FreeCAWH0vxz/usr/Mod/Tux/InitGui.py
Please look into the log file for further information
During initialization the error ""'BIMWorkbench21' already exists."" occurred in /home/marco/.local/share/FreeCAD/Mod/BIM021/./InitGui.py
Please look into the log file for further information
No module named 'Arch_rc'
No module named 'Arch_rc'Error: Initializing one or more of the Draft modules failed, Draft will not work as expected.
No module named 'Arch_rc'
The `.FCStd#` backup format is deprecated and may be removed in future versions.
To update, check the 'Preferences->General->Document->Use date and FCBak extension' option.
No module named 'Arch_rc'
Traceback (most recent call last):
  File "/home/marco/.local/share/FreeCAD/Mod/BIM.backup1724617972.7890484/./InitGui.py", line 435, in Activated
    from DraftGui import todo
  File "/tmp/.mount_FreeCAWH0vxz/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAWH0vxz/usr/Mod/Draft/DraftGui.py", line 48, in <module>
    import Draft
  File "/tmp/.mount_FreeCAWH0vxz/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAWH0vxz/usr/Mod/Draft/Draft.py", line 53, in <module>
    from draftutils.utils import ARROW_TYPES as arrowtypes
  File "/tmp/.mount_FreeCAWH0vxz/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAWH0vxz/usr/Mod/Draft/draftutils/utils.py", line 42, in <module>
    from draftutils import params
  File "/tmp/.mount_FreeCAWH0vxz/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAWH0vxz/usr/Mod/Draft/draftutils/params.py", line 31, in <module>
    import Arch_rc
  File "/tmp/.mount_FreeCAWH0vxz/usr/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

yorikvanhavre, 2024-09-24:
> File "/home/marco/.local/share/FreeCAD/Mod/BIM.backup1724617972.7890484/./InitGui.py"

There is something wrong there, that module shouldn't be loaded... Could you delete folder /home/marco/.local/share/FreeCAD/Mod/BIM.backup1724617972.7890484 manually? It seems like a leftover from an unfinished addon install...

marcosilveira2, 2024-09-25:
in FreeCAD 1.0RC works

luzpaz, 2024-10-16:
Can this ticket be closed in that case ?

closed 2024-10-17 by Roy-043