FreeCAD/FreeCAD #14541: Missing Materials directories breaks most functionality

morganrallen, 2024-06-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If no Materials directories are available; either through the system package or user .local FreeCAD will throw several high level errors, preventing many Workbenches from not loading.

When starting Part Design...
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'Base.FreeCADError'>: {'sclassname': 'N9Materials16MaterialNotFoundE', 'sErrMsg': 'FreeCAD Exception', 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': 'FreeCAD Exception', 'btranslatable': False, 'breported': False}
```

When accessing Materials Preferences...
```
Base exception thrown for 'MatGui::DlgSettingsDefaultMaterial'
<Exception> FreeCAD Exception
```

A normal user install should not be able to get into this state as the FreeCAD package will have installed the default Materials at the system level, I only ran into this situation by running locally built 0.22dev branches without any install steps, so I had neither the system nor in my `~/.local/share/FreeCAD/...`.

This is rather edge case but potentially could still be handled more gracefully in the unusual cases it could happen.

### Full version info

```shell
OS: Arch Linux (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37655 (Git)
Build type: Release
Branch: main
Hash: de9be04249c862891d2a6903b5661460452796c7
Python 3.12.3, Qt 6.7.1, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-07:
@davesrocketshop FYI
Is this a materials related issue or also the same as https://github.com/FreeCAD/FreeCAD/issues/6707
Related: https://github.com/FreeCAD/FreeCAD/issues/12844

krushia, 2024-07-16:
Testcase:

1. Open FreeCAD
2. Edit -> Preferences...
3. Material -> General
4. **Deselect** "Use built-in materials" (highlighted in screenshot below)
5. Close FreeCAD
6. Open just about anything in FreeCAD (Simple example file attached below if needed).

[cubesphere.FCStd.delete.this.extension.zip](https://github.com/user-attachments/files/16253797/cubesphere.FCStd.delete.this.extension.zip)

Errors will occur, and project will have severe data loss. Errors are not helpful. From example file the log contains:
```
Err: Cannot create object 'Box': (FreeCAD Exception)
Err: Cannot create object 'Sphere': (FreeCAD Exception)
```

[image]

davesrocketshop, 2024-07-16:
> @davesrocketshop FYI Is this a materials related issue or also the same as #6707 Related: #12844

This is not related to #6707 which is more of a corrupted install issue. That will always have problems. #12844 is a post 1.0 feature, and will not always prevent the problem.

The preference setting should probably go away. It was a legacy option that makes less sense with the new system. Built-in materials should always be available, except in rare cases such as unit testing.

luzpaz, 2024-07-18:
@davesrocketshop should this ticket be renamed in that case ?

davesrocketshop, 2024-07-19:
> @davesrocketshop should this ticket be renamed in that case ?

Not really. Assign it to me and label it as a blocker for 1.0

davesrocketshop, 2024-07-25:
@maxwxyz Please remove the blocker label. This will not be fixed for 1.0, but immediately post 1.0. All solutions impact translation or otherwise change behaviour too late in the game. As the original poster stated, this is a corner case and the issue can wait

maxwxyz, 2025-03-09:
Is this still an issue or already fixed?

davesrocketshop, 2025-03-09:
Still an issue

ezhukov, 2025-04-25:
Any workaround for this?

morganrallen, 2025-04-25:
off the top of my head, I think this should do the trick....

Preferences > Materials > User Directory should be filled out
and the directory its pointing at needs to exist.

ezhukov, 2025-04-25:
Thanks for a quick reply!
I was using `libfreecad-python3` in Debian to run my Python script and generate a beam geometry, but hit the above issue. Here is a list of files in that package https://packages.debian.org/sid/amd64/libfreecad-python3/filelist

Let me know if there are any possible workarounds/fixes you could think of!

luzpaz, 2025-11-02:
Hi @davesrocketshop, do you still want this ticket assigned to you ?

davesrocketshop, 2025-11-02:
yes

FEA-eng, 2026-09-30:
```
07:36:52  Cannot create object 'Box': (Material not found)
07:36:52  Cannot create object 'Sphere': (Material not found)
```

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260929 (Git shallow)
Build date: 2026/09/29 22:39:08
Build type: Release
Branch: grafted,grafted
Hash: 3160daf1e2b62b0472854da4848255c6b3ad554c
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```