FreeCAD/FreeCAD #14545: spread sheet not Undocked

Vincenzoferrara, 2024-06-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

in spreadsheet the "Undocked" and "doked" function is disabled.

another bug i think related, in:
view > document window > fullscreen and disabled
but view > fullscreen and enabled

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git) AppImage
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT) [ OS: Italian/Italy (it_IT) ]
Installed mods: 
  * AutoOpenSpreadsheet
  * ProDarkThemePreferencePack 1.0.0
  * CommandPanel
  * IconThemes
  * fasteners.backup1715436808.37841 0.5.16 (Disabled)
  * FeedsAndSpeeds 0.5.0
  * ExplodedAssembly
  * OpenTheme 2024.5.16
  * Freecad-Built-in-themes-beta 1.2.2
  * fasteners 0.5.21
  * TabBar
  * Assembly4 0.50.12
  * CubeMenu
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-07:
Can you please add steps how to reproduce this and a screenshot?
I do not understand the issue.

Vincenzoferrara, 2024-06-08:
https://github.com/FreeCAD/FreeCAD/assets/64591523/de00d2b2-4cd1-4030-a96e-b1457aaa0bae

my freecad is in italian language.

moreover you can't take screenshots when the curtain is down, it doesn't take any shotcuts

maxwxyz, 2024-06-08:
Only 3D views can be undocked https://wiki.freecad.org/Std_ViewDockUndockFullscreen

Vincenzoferrara, 2024-06-09:
i'm using this macro for Undocked

`from PySide2 import QtWidgets, QtCore
import FreeCADGui as Gui

class WinSplitter(QtCore.QObject):

    class SplitWindow(QtWidgets.QMainWindow):

        def __init__(self, wid, res, title):
            super(WinSplitter.SplitWindow, self).__init__()
            self.wid = wid
            self.res = res
            self.setCentralWidget(self.wid)
            self.setWindowTitle(title)
            self.show()

        def closeEvent(self, ev):
            self.res.layout().addWidget(self.wid)
            return super(WinSplitter.SplitWindow, self).closeEvent(ev)

    def __new__(cls):
        if not hasattr(Gui.getMainWindow(), 'winsplitter'):
            return super().__new__(cls)
        else:
            return None

    def __init__(self):
        super(WinSplitter, self).__init__()
        Gui.getMainWindow().centralWidget().findChild(QtWidgets.QTabBar).tabBarDoubleClicked.connect(self.onTBDC)
        setattr(Gui.getMainWindow(), 'winsplitter', self)
        self.splitwin = []

    def onTBDC(self, idx):
        if idx >= 0:
            sw = Gui.getMainWindow().centralWidget().subWindowList()[idx]
            self.splitwin.append(WinSplitter.SplitWindow(sw.widget(), sw, sw.windowTitle()))

WinSplitter()`

WandererFan, 2024-06-10:
When something works with 3d view and not in Spreadsheet and TechDraw, it is usually because there is code checking against View3dInventor that should be checking for the  base class MDIView.  

Spreadsheet and TechDraw windows are derived from MDIView, as is View3dInventor.

PR #22544 (2025-07-24): Gui: Allow undocking and fullscreen for all MDI windows
Split from #22204 as that PR evolved out of the original scope. The discussion over there might still be relevant.

Currently the 3D viewer can be undocked and set to fullscreen, either via shortcut ("V, U" or "F11") or via "View" -> "Document window". This PR enables this functionality for all MDI widgets (Spreadsheet, TechDraw, Dependency Graph, Start Page, ...)

Edit:
closes #14545