FreeCAD/FreeCAD #14546: add rotation is moved for all axes

Vincenzoferrara, 2024-06-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

when I use the fixed gimbal, I want to be able to change the inclination and also its position for each axis.

I have set the distance in a spreadsheet, so if it were possible I would like to have this functionality too.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git) AppImage
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT) [ OS: Italian/Italy (it_IT) ]
Installed mods: 
  * AutoOpenSpreadsheet
  * ProDarkThemePreferencePack 1.0.0
  * CommandPanel
  * IconThemes
  * fasteners.backup1715436808.37841 0.5.16 (Disabled)
  * FeedsAndSpeeds 0.5.0
  * ExplodedAssembly
  * OpenTheme 2024.5.16
  * Freecad-Built-in-themes-beta 1.2.2
  * fasteners 0.5.21
  * TabBar
  * Assembly4 0.50.12
  * CubeMenu
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-06-07:
try using STD_Placement

https://github.com/FreeCAD/FreeCAD/assets/32273833/545cf49c-f5e7-48cb-9181-9e56ef0fe23c

closed 2024-06-07 by maxwxyz

Vincenzoferrara, 2024-06-07:
this in workbanch assembly