FreeCAD/FreeCAD #14550: AppImage: Path workbench: FCStd stores absolute path to toolbit Shapefile

jes, 2024-06-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using the AppImage (which mounts at something like `/tmp/.mount_FreeCAUR2vXf`), if I create a Path job and set up a toolbit, and then close FreeCAD and reopen FreeCAD, and then try to edit the toolbit by double-clicking it in the model tree, it says "Toolbit cannot be edited: Shapefile not found".

The reason is that FreeCAD stores the absolute path to the Shapefile inside the FCStd, which is now wrong because when I reopened FreeCAD it got mounted at a different path under `/tmp`.

### Full version info

```shell
[code]
OS: Ubuntu 20.04.5 LTS (Regolith:GNOME-Flashback:GNOME/regolith)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Curves 0.4.2
  * Render 2022.4.0
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

A workaround is to run `mount` and see where FreeCAD has been mounted, and use the property editor to change the path in the "Bit shape" property to point to the new directory, and then the bit can be edited.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-07:
Can you confirm that is the same issue than https://github.com/FreeCAD/FreeCAD/issues/6707 or is it only happening in CAM?
Related: https://github.com/FreeCAD/FreeCAD/issues/12844

jes, 2024-06-07:
I expect it is broadly the same problem.

maxwxyz, 2024-06-15:
I'll close this then in favor of #6707

closed 2024-06-15 by maxwxyz