FreeCAD/FreeCAD #14553: Part.Face.curveOnSurface return wrong result

tomate44, 2024-06-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It seems that Part.Face.curveOnSurface is broken.
It always return a pcurve, even when it shouldn't.
```
cyl = Part.makeCylinder(2, 10)
f = cyl.Face2
e = cyl.Edge3
Part.show(f)
Part.show(e)
f.curveOnSurface(e)
# return (<Circle2d object>, 0.0, 6.283185307179586)
# while it should (and used to) return None
```

### Full version info

```shell
OS: Manjaro Linux (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37665 (Git)
Build type: Unknown
Branch: main
Hash: 62463fa0bd2c6962550ef67f94ca527b81b21cd9
Python 3.12.3, Qt 6.7.1, Coin 4.0.2, Vtk , OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * lattice2 1.0.0
  * CurvedShapes 1.0.5
  * casat
  * sheetmetal 0.4.3
  * OpenTheme 2024.4.20
  * CurvesWB 0.6.36
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-07:
@bgbsww related to TNP?

bgbsww, 2024-06-07:
Quick test shows it happening in 0.21.2 where TNP is still off.  It also happens in LS3 and in 0.20.2  Is the example code correct, and on what version exactly did it return None?

tomate44, 2024-06-09:
Indeed, I observe the same behavior on 0.21.2, and on Appimage 0.20.2 with OCCT 7.6.3
That look strange because I use this function for years, and I am sure it worked fine in the past.
So I have no clue what's going on ???
By the way, here is another script showing the bug : 
```
cone = Part.makeCone(2, 4, 10)
seam = cone.Edge2
Part.show(seam, "Seam")
# This edge has 2 pcurves
print(seam.curveOnSurface(0))
print(seam.curveOnSurface(1))
print(seam.curveOnSurface(3))  # None = no more pcurves

topface = cone.Face2
Part.show(topface, "TopFace")

cos = topface.curveOnSurface(seam)
print(cos)
# should be None because seam has no pcurve on the top face
# instead, the returned curve2D is a projection of the seam on the topface
edge = cos[0].toShape(topface.Surface, cos[1], cos[2])
Part.show(edge, "PCurve")

```

tomate44, 2024-06-10:
I checked into Opencascade source code.
When Face is a plane and no existing pcurve is found, curveOnSurface() does a [projection on the plane](https://git.dev.opencascade.org/gitweb/?p=occt.git;a=blob;f=src/BRep/BRep_Tool.cxx;hb=1103eb60af84b32f44af1b1fc4222f32e4575daf#l339).
Quite strange it does not check if Edge and Face are coplanar beforehand.
Anyway, I was obviously wrong.
Apologies for the wasted time. I close.

closed 2024-06-10 by tomate44