FreeCAD/FreeCAD #14557: Path WB fails to save a file when postprocessing

InnerBushman, 2024-06-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When postprocessing a job in Path workbench there is an error:
```  File "/usr/lib/freecad/Mod/Path/Path/Post/scripts/linuxcnc_post.py", line 293, in export
    gfile = pythonopen(filename, "w")
            ^^^^^^^^^^
name 'pythonopen' is not defined```

### Full version info

```shell
OS: Manjaro Linux (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.12.3, Qt 6.7.0, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

CAM/Path

### Anything else?

Suggested solution:

```--- ./linuxcnc_post_old.py	2024-06-07 19:15:13.605470705 +0200
+++ linuxcnc_post.py	2024-06-07 19:13:42.475699518 +0200
@@ -129,7 +129,7 @@
 TOOL_CHANGE = """"""
 
 # to distinguish python built-in open function from the one declared below
-if open.__module__ in ["__builtin__", "io"]:
+if open.__module__ in ["__builtin__", "_io"]:
     pythonopen = open```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-07:
PLease add steps how to reproduce this, if possible with a demo file.
Could your try your document in the latest dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

InnerBushman, 2024-06-08:
Hey Max,
I've updated the post with steps to reproduce and uploaded an example file.
I also tested it with dev build (appimage) as you suggested, but quickly realized it won't matter, as it's on different python version (but yes, it does work fine in dev).

maxwxyz, 2024-06-08:
I'll  close it then if the issue is not reproducible in the dev.

InnerBushman, 2024-06-08:
I cannot reproduce it as I run it from appimage. I'm sure if someone would run the dev version on python 3.12, it can be reproduced.

maxwxyz, 2024-06-08:
@adrianinsaval what version will 1.0 have?

InnerBushman, 2024-06-08:
After some investigation I've came to the conclusion it is fixed in next release.
The change that is fixing it is removing the check altogether [here](https://github.com/FreeCAD/FreeCAD/commit/ff11cd25ee99b5e959616f3805f0d9bfd3a0be76)
There still remains the fact it is still broken in 0.21 due to some changes in Python 3.12 that I don't completely undestand:
[https://docs.python.org/3/whatsnew/3.12.html#io](https://docs.python.org/3/whatsnew/3.12.html#io)
Since I have a working solution, and it's actually fixed in next version, I'll leave it to your discretion to close the issue. Although I would ask if the change could be implemented in the current version as a bug fix.

maxwxyz, 2024-06-08:
current version will not get a bug fix as we are already in feature freeze for the next version.

closed 2024-06-08 by maxwxyz

InnerBushman, 2024-06-08:
It's a bug, not a feature :D