FreeCAD/FreeCAD #14559: [Problem] New unified measurement tool doesn't work properly in the Integrated Assembly WB

MiqCG, 2024-06-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to measure (eg. distances, angles...) between two bodies of an assembly file or even trying to measure over a body inside the assembly, the results are incorrect.
The old distance measuring tool (in the Edit menu) works in those situations.

I have searched in Github with [is:issue is:open measure label:"WB Assembly" ], with no results.

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37668 (Git) AppImage
Build type: Release
Branch: main
Hash: ea68c5e88cdc80044ede482f6ad908820f4c1c82
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * freecad.gears 1.2.0
  * A2plus 0.4.64a
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * Trails 2022.1.0
  * PieMenu 1.2.7
  * fasteners 0.5.21
  * Render 2024.1.23
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-08:
Please provide a demo file or screenshots how to reproduce this.

MiqCG, 2024-06-08:
Looks

> Please provide a demo file or screenshots how to reproduce this.

Done. Sorry

This demo has been made coping bodies inside the assembly file but happens the same with inserting files with the insert component tool of the Assembly WB.

maxwxyz, 2024-06-08:
@hlorus @PaddleStroke any idea why the measurement fails in assemblies?

WandererFan, 2024-06-08:
Sample does not load.  Missing link targets for 210-005 Barra compas and 210-007 Barra corredora-compàs.

09:42:39  6.1e-08 <App> Document.cpp(2205): Ass_test_Github#_10_005_Barra_compas.LinkedObject: Link not restored
Linked object: Body
Linked file: #210-005 Barra compàs.FCStd
09:42:39  0.00010554 <App> Document.cpp(2205): Ass_test_Github#_10_007_Barra_corredora_comp__s.LinkedObject: Link not restored
Linked object: Body
Linked file: #210-007 Barra corredora-compàs.FCStd

MiqCG, 2024-06-08:
> Sample does not load.

Sorry, I get lost in between several tests and I attached a file with linked bodies. Now I have changed it with one with copied bodies inside. Almost the same (bad) results but take into consideration that screencasts were made with the linked bodies one.

furgo16, 2024-06-16:
I just noticed too that assemblies cannot be measured with the new measurement tool

Steps to reproduce:

1. Open attached file with FreeCAD: [SimpleAsy.zip](https://github.com/user-attachments/files/15857539/SimpleAsy.zip)
1. Double-click on the Assembly object to activate the Assembly workbench
2. Click on the measure tool
3. Click on one edge of the cube
4. Expected: a measurement is shown (10 mm)
5. Actual: no measurement is shown

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37731 (Git) Snap 1052
Build type: Release
Branch: main
Hash: 6623b605282aa5ae61ae3df0adff2f6c857477fa
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * OpenTheme 2024.5.16
  * Curves 0.6.39
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.5
  * Ondsel-Lens 2024.5.6.01
  * Assembly4 0.50.13
```

MiqCG, 2024-08-03:
I am still experiencing bad results with revision 38385, both with assemblies using links and assemblies using inserted parts with the assembly tool.
I attach a folder with some simple test files. The measures are incorrect and the dimension lines are wrongly positioned.
If you need that I create a new issue, just tell me.

```
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git) AppImage
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
```

[Measure.zip](https://github.com/user-attachments/files/16482795/Measure.zip)

maxwxyz, 2024-08-04:
@MiqCG at least the measurements are correct with your file.
It is displayed at a wrong location, but this is this issue: https://github.com/FreeCAD/FreeCAD/issues/13718

MiqCG, 2024-08-04:
Which measurements are correct? On my side the tool only work measuring the dimensions of one part.
Did you tried to measure between different parts?
See the attached video.

https://github.com/user-attachments/assets/1761b17b-b4c5-4b1d-ab12-26288ac08f8f

MiqCG, 2024-09-10:
And it is still not working on 1.0.0RC1 (38643).

1. Measure tool measures and displays correctly only the parts that are grounded at the same place they have been inserted (no transformation before). 
2. The measures between locations of the same part are correct but displayed incorrectly if the part is not at the origin. 
3. The dimensions between two joined parts (assembled) are incorrect in value and placement.

I attach a simple example with 3 different parts newly created and an assembly file. You can check, for example, that moving the slider and measuring 2 points of the prisms results in the same dimension every time.

[This](https://github.com/FreeCAD/FreeCAD/pull/14918) PR hasn't solved the issue.
@maxwxyz , @luzpaz , I ask for reopening this issue.

[Measure.zip](https://github.com/user-attachments/files/16947357/Measure.zip)

`
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git) AppImage
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * fasteners 0.5.27
`

blumberg, 2024-09-15:
It seams like the Measurement tool is looking into the local coordinate of the part and not the global assembly coordinate, thus giving the wrong dimension.

If you measure the position of a point, you can see it shows the local position.

PaddleStroke, 2024-09-16:
I would guess that the measurement tool (and quick measure as well btw) are not using the global placement of object. The core getGlobalPlacement was not handling links correctly. So in assembly I made a getGlobalPlacement that works with links.

And this function has been moved into core recently : https://github.com/FreeCAD/FreeCAD/pull/15262#issuecomment-2350075307

maxwxyz, 2024-09-16:
@hlorus FYI

PaddleStroke, 2024-09-16:
I will fix in quickmeasure, then will have a look at the UMT if @hlorus doesn't do it before.

hlorus, 2024-09-16:
@PaddleStroke there's still #16251 open which should solve it for UMF.

maxwxyz, 2024-09-16:
@hlorus #16251 got merged now

PaddleStroke, 2024-09-16:
Ah good, then only quick measure needs to be fixed

hlorus, 2024-09-17:
@PaddleStroke Measurements still don't get recomputed automatically when parts in an assembly are dragged, like they do otherwise (e.g. if the placement is changed with the placement dialog).

PaddleStroke, 2024-09-17:
https://github.com/FreeCAD/FreeCAD/pull/16601 is fixing quickMeasure.

Now this issue should perhaps not be closed since @hlorus report this new issue. I'm not sure why this would happen. Sounds like a recompute or purgeTouched issue.

chennes, 2024-09-20:
Can we close this issue and open a more focused one? "Doesn't work properly" is too broad to be actionable, and the new issue might not be a blocker anymore.

MiqCG, 2024-09-20:
I suspect that you are not asking this to me, but if it helps I will explain that "doesn't work properly" was a way of saying "the measures are incorrect". Nothing else, sorry for the ambiguity. 
So, if the result of the bug that hlorus has pointed out is the same (incorrect measures), I can change the title without changing the spirit of the initial issue.

chennes, 2024-09-20:
"Doesn't work properly" was totally fine (and true!) when you started the issue. My point is that now it *mostly* does work properly, and there is a narrow set of circumstances when it fails. So to avoid this issue turning into a "catch all" for any measurement-in-assembly bugs, I am proposing that we proactively close it and create more narrowly-defined reports.

maxwxyz, 2024-10-12:
What is left to close this issue? In which cases is the measurement wrong?

MiqCG, 2024-10-12:
I have made some tests considering the message of @hlorus 
> Measurements still don't get recomputed automatically when parts in an assembly are dragged, like they do otherwise (e.g. if the placement is changed with the placement dialog).

I have tested Distance, Angle and Distance free measurements. Distance and Angle do  update correctly if you do a recompute (Ctrl - r) of the assembly as indicated in the tree icon. I don't know if the expected measurement behaviour is an automatic update after dragging. Anyway, **Distance free doesn't update** , even with a recompute.

On the other hand, I think I have discovered that the label of Angle measurement gets printed far away when values are nearer to 180º. See end part of attached screencast.

Tests made with 1.1.0-38898. 
[Measure_drag.zip](https://github.com/user-attachments/files/17349440/Measure_drag.zip)

https://github.com/user-attachments/assets/6d5e3716-b28e-48af-9c25-5f6a2b54670c

maxwxyz, 2024-10-14:
@hlorus ^

hlorus, 2024-10-14:
Distance Free isn't expected to update since it just shows the distance between two coordinates and isn't linked to any object/element.

Since measurments update correctly when a recompute is triggered manually i suspect that dragging elements simply doesn't trigger a full recompute due to performance reasons. @PaddleStroke?

PaddleStroke, 2024-10-14:
Yes indeed. But I'm not an expert of the recompute mechanism. Perhaps we can do a recompute without loosing too much performances?

MiqCG, 2024-10-14:
> Perhaps we can do a recompute without loosing too much performances?

IMHO this could be a good default, and maybe it can come with a Preference option to disable it, in case it turns out annoying with big assemblies. 

On the other hand, in Assembly WB I do recompute a lot - every time I see the white/blue check icon - that is when making active an Assembly container, after creating a new joint, after modifying a joint... Maybe is not necessary to recompute an then the icon is misleading, but if it is a convenient action, I propose to program an automatic recompute in this situations too (or on some of them).

chennes, 2024-10-14:
@maxwxyz can you close this issue and create a new more focused one?

PR #14918 (2024-07-21): Measure: Fix measure issues in assemblies
As described in #14559 measurements don't properly work with the integrated assembly workbench. This PR adds checks for App::Link types and resolves them to get the responsible module and fixes the retrieval of the located shapes.

Closes #14559
closed 2024-10-14 by maxwxyz