FreeCAD/FreeCAD #14562: Crash when input 0 in Scale property of App::Link

Jolbas, 2024-06-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Running this python from console leads to a crash

```python
doc = App.newDocument()
box = doc.addObject("Part::Box","Box")
link = doc.addObject('App::Link','Link')
link.setLink(box)
Gui.Selection.addSelection(link)
link.Scale = 0
doc.recompute()
```

or just select any link object, edit its Scale property and type "0"

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37439 (Git)
Build type: Release
Branch: main
Hash: 202aa359b3476a9e1905be8328c05e54482fe231
Python 3.11.7, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Swedish/Sweden (sv_SE)
Installed mods: 
  * Curves.backup1717858610.556698 0.6.8 (Disabled)
  * kugghjul.py
  * OpenDark 2024.3.13
  * dxf-library
  * fcgear 1.0.0
  * __pycache__
  * Manipulator 1.5.0
  * sheetmetal 0.2.61
  * tida.py
  * Curves 0.6.38

0.21.1 doesn't crash
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-06-08:
Instead of crashing, _FreeCAD_ reports:

> Unhandled unknown exception caught in GUIApplication::notify.

```ruby
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37665 (Git)
Build type: Debug
Branch: main
Hash: 62463fa0bd2c6962550ef67f94ca527b81b21cd9
Python 3.10.12, Qt 5.15.14, Coin 4.0.2, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

Jolbas, 2024-06-08:
Call stack from crash report:

```
Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x19162aa60 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x191662c20 pthread_kill + 288
2   libsystem_c.dylib             	       0x19156fa30 abort + 180
3   libc++abi.dylib               	       0x191619d08 abort_message + 132
4   libc++abi.dylib               	       0x191609fc0 demangling_terminate_handler() + 348
5   libobjc.A.dylib               	       0x1912a81e0 _objc_terminate() + 160
6   libc++abi.dylib               	       0x1916190cc std::__terminate(void (*)()) + 16
7   libc++abi.dylib               	       0x19161c348 __cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) + 88
8   libc++abi.dylib               	       0x19161c28c __cxa_throw + 308
9   libTKMath.7.7.2.dylib         	       0x11cab5f4c gp_Trsf::SetValues(double, double, double, double, double, double, double, double, double, double, double, double) + 548
10  Part.so                       	       0x11c5ffaf0 Part::TopoShape::_makeTransform(Part::TopoShape const&, Base::Matrix4D const&, char const*, bool, bool) + 292
11  Part.so                       	       0x11c5ff978 Part::TopoShape::transformShape(Base::Matrix4D const&, bool, bool) + 112
12  Part.so                       	       0x11c3b4c1c _getTopoShape(App::DocumentObject const*, char const*, bool, Base::Matrix4D*, App::DocumentObject**, bool, bool, std::__1::set<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>, App::DocumentObject const*) + 6884
13  Part.so                       	       0x11c3b02e4 Part::Feature::getTopoShape(App::DocumentObject const*, char const*, bool, Base::Matrix4D*, App::DocumentObject**, bool, bool, bool) + 736
14  PartGui.so                    	       0x16601ffdc PartGui::hasShapesInSelection() + 132
15  PartGui.so                    	       0x16602bb04 CmdCheckGeometry::isActive() + 12
16  libFreeCADGui.dylib           	       0x105b656f8 Gui::Command::testActive() + 544
17  libFreeCADGui.dylib           	       0x105b71e14 Gui::CommandManager::testActive() + 60
18  libFreeCADGui.dylib           	       0x105f4e248 Gui::MainWindow::_updateActions() + 788
19  libQt5Core.5.15.13.dylib      	       0x107e69e84 0x107d24000 + 1334916
20  libQt5Core.5.15.13.dylib      	       0x107e4d094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
21  CoreFoundation                	       0x19174246c __CFRunLoopDoSource0 + 176
22  CoreFoundation                	       0x1917421dc __CFRunLoopDoSources0 + 244
23  CoreFoundation                	       0x191740dc8 __CFRunLoopRun + 828
24  CoreFoundation                	       0x191740434 CFRunLoopRunSpecific + 608
25  HIToolbox                     	       0x19bee419c RunCurrentEventLoopInMode + 292
26  HIToolbox                     	       0x19bee3fd8 ReceiveNextEventCommon + 648
27  HIToolbox                     	       0x19bee3d30 _BlockUntilNextEventMatchingListInModeWithFilter + 76
28  AppKit                        	       0x194f9fd68 _DPSNextEvent + 660
29  AppKit                        	       0x195795808 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
30  AppKit                        	       0x194f9309c -[NSApplication run] + 476
31  libqcocoa.dylib               	       0x1091444d0 0x109118000 + 181456
32  freecad                       	       0x1040d729c main + 3356
33  dyld                          	       0x1912da0e0 start + 2360
```

commit 0290a875 (2024-06-17): Part: Handle zero scale in TopoShape::_makeTransform

With this commit the TopoShape::_makeTransform() use makeGTransform() if scale is zero to avoid crash. Fixes #14562
closed 2024-06-17 by chennes