FreeCAD/FreeCAD #14563: [Toponaming] Opening a PartDesign sketch with invalid external geometry crashes FreeCAD

jffmichi, 2024-06-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

Create Sketch A with some geometry
Create Sketch B and reference external geometry from Sketch A
Delete referenced geometry in Sketch A
Open Sketch B
-> Sketch Task Panel can't be closed and FreeCAD crashes after a few seconds (probably when moving the mouse somwhere in the main view

See attached file and open Sketch001

### Full version info

```shell
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37438 (Git)
Build type: Release
Branch: main
Hash: 58083879b765846daaeab4fdb125e418138e5874
Python 3.12.3, Qt 6.7.1, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * kicadStepUpMod 11.1.7
  * dodo 1.0.1
  * sheetmetal 0.4.16
  * fcgear 1.0.0
  * fasteners 0.5.21
  * Assembly4 0.50.12
  * Curves 0.6.37
  * lattice2 1.0.0
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-06-09:
Here is the backtrace.

<details><summary>Backtrace</summary>

```lua
Process 1731 stopped
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x0)
    frame #0: 0x00007fff93b06937 SketcherGui.so`SketcherGui::EditModeGeometryCoinManager::updateGeometryColor(Sketcher::GeoListModel<std::unique_ptr<Sketcher::GeometryFacade const, std::default_delete<Sketcher::GeometryFacade const>>> const&, bool) [inlined] std::vector<int, std::allocator<int>>::operator[](__n=0, this=size=0) at stl_vector.h:1126:39
   1123       operator[](size_type __n) _GLIBCXX_NOEXCEPT
   1124       {
   1125         __glibcxx_requires_subscript(__n);
-> 1126         return *(this->_M_impl._M_start + __n);
   1127       }
   1128
   1129       /**
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: address not mapped to object (fault address: 0x0)
  * frame #0: 0x00007fff93b06937 SketcherGui.so`SketcherGui::EditModeGeometryCoinManager::updateGeometryColor(Sketcher::GeoListModel<std::unique_ptr<Sketcher::GeometryFacade const, std::default_delete<Sketcher::GeometryFacade const>>> const&, bool) [inlined] std::vector<int, std::allocator<int>>::operator[](__n=0, this=size=0) at stl_vector.h:1126:39
    frame #1: 0x00007fff93b0692c SketcherGui.so`SketcherGui::EditModeGeometryCoinManager::updateGeometryColor(Sketcher::GeoListModel<std::unique_ptr<Sketcher::GeometryFacade const, std::default_delete<Sketcher::GeometryFacade const>>> const&, bool) [inlined] SketcherGui::CoinMapping::getPointGeoId(layerindex=0, pointindex=0, this=0x000055555ac10078) at EditModeCoinManagerParameters.h:488:53
    frame #2: 0x00007fff93b0691a SketcherGui.so`SketcherGui::EditModeGeometryCoinManager::updateGeometryColor(this=0x0000555555e47a40, geolistfacade=0x00007fffffffafa0, issketchinvalid=false) at EditModeGeometryCoinManager.cpp:195:50
    frame #3: 0x00007fff93b01f02 SketcherGui.so`SketcherGui::EditModeCoinManager::updateColor(this=0x000055555ac0fde0, geolistfacade=0x00007fffffffafa0) at EditModeCoinManager.cpp:884:54
    frame #4: 0x00007fff93b01fc5 SketcherGui.so`SketcherGui::EditModeCoinManager::updateColor(this=0x000055555ac0fde0) at EditModeCoinManager.cpp:877:16
    frame #5: 0x00007fff93b1cf5f SketcherGui.so`SketcherGui::ViewProviderSketch::updateColor(this=0x000055555abeffb0) at ViewProviderSketch.cpp:2607:33
    frame #6: 0x00007fff93b2165b SketcherGui.so`SketcherGui::ViewProviderSketch::onSelectionChanged(this=0x000055555abeffb0, msg=0x00007fffffffb6a0) at ViewProviderSketch.cpp:2226:30
    frame #7: 0x00007ffff71ced1f libFreeCADGui.so`Gui::SelectionObserver::_onSelectionChanged(this=<unavailable>, msg=<unavailable>) at Selection.cpp:152:27
    frame #8: 0x00007ffff71dd5cb libFreeCADGui.so`void std::__invoke_impl<void, void (Gui::SelectionObserver::*&)(Gui::SelectionChanges const&), Gui::SelectionObserver*&, Gui::SelectionChanges const&>((null)=__invoke_memfun_deref @ 0x00007fffffffb300, __f=<unavailable>, __t=<unavailable>, (null)=<unavailable>) at invoke.h:74:46
    frame #9: 0x00007ffff71dd5e9 libFreeCADGui.so`boost::detail::function::void_function_obj_invoker1<std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>, void, Gui::SelectionChanges const&>::invoke(boost::detail::function::function_buffer&, Gui::SelectionChanges const&) [inlined] std::__invoke_result<void (Gui::SelectionObserver::*&)(Gui::SelectionChanges const&), Gui::SelectionObserver*&, Gui::SelectionChanges const&>::type std::__invoke<void (Gui::SelectionObserver::*&)(Gui::SelectionChanges const&), Gui::SelectionObserver*&, Gui::SelectionChanges const&>((null)=0x00007fffffffb6a0, (null)=<unavailable>, __fn=<unavailable>) at invoke.h:96:40
    frame #10: 0x00007ffff71dd5e4 libFreeCADGui.so`boost::detail::function::void_function_obj_invoker1<std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>, void, Gui::SelectionChanges const&>::invoke(boost::detail::function::function_buffer&, Gui::SelectionChanges const&) [inlined] void std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>::__call<void, Gui::SelectionChanges const&, 0ul, 1ul>((null)=<unavailable>, __args=0x00007fffffffb308, this=<unavailable>) at functional:506:24
    frame #11: 0x00007ffff71dd5e0 libFreeCADGui.so`boost::detail::function::void_function_obj_invoker1<std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>, void, Gui::SelectionChanges const&>::invoke(boost::detail::function::function_buffer&, Gui::SelectionChanges const&) [inlined] void std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>::operator()<Gui::SelectionChanges const&, void>((null)=0x00007fffffffb6a0, this=<unavailable>) at functional:591:32
    frame #12: 0x00007ffff71dd5dc libFreeCADGui.so`boost::detail::function::void_function_obj_invoker1<std::_Bind<void (Gui::SelectionObserver::* (Gui::SelectionObserver*, std::_Placeholder<1>))(Gui::SelectionChanges const&)>, void, Gui::SelectionChanges const&>::invoke(function_obj_ptr=<unavailable>, a0=0x00007fffffffb6a0) at function_template.hpp:158:11
[... cut ...]
    frame #83: 0x00007ffff3229e40 libc.so.6`__libc_start_main_impl(main=(FreeCAD`main at MainGui.cpp:107:1), argc=1, argv=0x00007fffffffdd58, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdd48) at libc-start.c:392:3
    frame #84: 0x0000555555569c05 FreeCAD`_start + 37
(lldb)
```

```ruby
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37665 (Git)
Build type: Debug
Branch: main
Hash: 62463fa0bd2c6962550ef67f94ca527b81b21cd9
Python 3.10.12, Qt 5.15.14, Coin 4.0.2, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

</details>

bgbsww, 2024-06-15:
Thanks for this, it is definitely an issue.  Now looking at it.

PR #14741 (2024-06-17): Toponaming: Fix crash with invalid external geometry
Fix #14563.   Could not devise a test to trigger this fail, which comes as a result of mouse movement triggering a redraw of a sketch in edit mode that contains invalid external geometry because the validation when the sketch goes into edit mode was failing to catch the correct exception type thrown by TNP code.
closed 2024-06-17 by chennes