FreeCAD/FreeCAD #14564: Arch/Bim Wall, Colinear wall segment / sketch edge with opposite direction / Left-Right Align resulted in wall segment corner touching corner - this is not acceptable in the OCC engine.

BanceFC, 2024-06-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After modifying a Sketcher polyline, (by adding to it later,) Arch wall tool fails in Left/Right mode (Centered works.)

I gather this is caused by adjoining at the corner. Strangely the Wall tool works if the addition is reversed (in direction.)

Forum topic:-https://forum.freecad.org/viewtopic.php?t=88295

Maybe related to https://github.com/FreeCAD/FreeCAD/issues/11957

FCStd File (renamed to .TXT.)

[ProblemSketch .TXT](https://github.com/user-attachments/files/15750703/ProblemSketch.TXT)

### Full version info

```shell
OS: Manjaro Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37543 (Git) AppImage
Build type: Release
Branch: main
Hash: 3eb45b045cdda9d5bfe11baba2567a1df90f064c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Curves 0.6.29
  * fasteners 0.5.15
  * freecad.gears 1.2.0
  * dodo 1.0.1
  * A2plus 0.4.64
  * sheetmetal 0.4.5
  * Assembly4 0.50.12
  * fcgear 1.0.0
  * AirPlaneDesign 0.4.0
  * CurvedShapes 1.0.8
  * DynamicData 2.60.0
  * kicadStepUpMod 11.1.0
  * Plot 2022.4.17
  * Nodes 0.1.36
  * BOLTSFC 2022.11.5
```

### Subproject(s) affected?

None

### Anything else?

https://forum.freecad.org/viewtopic.php?t=88295

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

paullee0, 2024-06-09:
This happens in cornercase when there are 2 consecutive wall segment (i.e. 2 consecutive sketch's edges) are colinear, but either have opposite Align (Left/Right) or Sketch Edge direction are opposite but their Align are not Center, this results in 2 consecutive Wall segment volume just touching themselves at a corner.  This is not acceptable to OCC and results in errror.

Needs to see if there is algorithm to reveal this cornercase, either make a warning, or prevent this happens, or otherwise.

paullee0, 2024-06-30:
> Needs to see if there is algorithm to reveal this cornercase, either make a warning, or prevent this happens, or otherwise.

OK, would see if this idea works :  When 2 consecutive wall segment have opposite Align (or similar case when the Orientation of the edges are opposite), the face of the particular segment is not generated.

paullee0, 2024-07-01:
> OK, would see if this idea works : When 2 consecutive wall segment have opposite Align (or similar case when the Orientation of the edges are opposite), the face of the particular segment is not generated.

Study at https://forum.freecad.org/viewtopic.php?p=768871#p768871 :)

paullee0, 2024-07-10:
[Pull Request](https://github.com/FreeCAD/FreeCAD/pull/15350)  :)

luzpaz, 2024-07-28:
PR was closed because it was not the correct fix

paullee0, 2024-07-29:
> PR was closed because it was not the correct fix

Not sure I understand, which PR you are referring to?  closed?  Thanks :)

yorikvanhavre, 2024-07-30:
Closed with #15530

paullee0, 2024-07-30:
This #14564 is in fact to be fixed by #15350, the latter of which IMHO should be reviewed and merged soon :)

yorikvanhavre, 2024-07-31:
Ok I'll reopen and attribute to #15350

closed 2024-08-08 by yorikvanhavre