FreeCAD/FreeCAD #14565: Crashes when loading NUC12STEP file (linked)

audunmg, 2024-06-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Crashes when loading NUC12STEP  from here: https://www.intel.com/content/www/us/en/support/articles/000006820/intel-nuc.html

Sometimes it crashes on it's own, sometimes it's hanging with 100% CPU usage, but most of the time, CPU usage increases to 100% of one core and memory usage is increasing until OOM at around 60GB.

Since the file is freely available from the Intel link above, this is reproducible.

### Full version info

```shell
[code]
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.12.3, Qt 6.7.0, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
[/code]
```

### Subproject(s) affected?

None

### Anything else?

Thank you for FreeCAD!

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-11:
can confirm, also on the latest weekly
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37669 (Git)
Build type: Release
Branch: main
Hash: 3b4598ce04d7aaec69feeb560c29ae18f0236e0c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CurvedShapes 1.0.9
  * Curves 0.6.36
  * fasteners 0.5.21
  * OpenTheme 2024.5.16
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.14

```

luzpaz, 2025-07-03:
@audunmg which file specifically ?

FEA-eng, 2026-09-28:
This model is huge, but it was imported eventually:

[image]

Maybe it's also OS-dependent.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260922 (Git shallow)
Build date: 2026/09/22 20:26:30
Build type: Release
Branch: (HEAD detached at 019f5c50a)
Hash: 019f5c50a945743b89cabf422ac5f5f1e297d890
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```