FreeCAD/FreeCAD #14566: The visibility icon needs to work with more styles

NomAnor, 2024-06-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is a follow up of #14342 and PR #14393. The visibility icon should work in all themes.

@kadet1090 [proposed](https://github.com/FreeCAD/FreeCAD/pull/14393#issuecomment-2155662668) using single color icons that are easier to style. But they don't look good in overlay.

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37330 (Git)
Build type: Debug
Branch: pd-icons
Hash: a094bf0df7e2b3e5c5b2e3b158a9040a6474892f
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.3.0
Locale: German/Germany (de_DE)
Installed mods: 
  * Manipulator 1.5.7
  * ThreadProfile 1.84.0
  * DynamicData 2.62.0
  * Assembly3.backup1715897154.0970206 0.11.3 (Disabled)
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * freecad.gears 1.2.0
  * fasteners 0.4.21
```

### Subproject(s) affected?

Core

### Anything else?

Here are some example screenshots with different styles. More screenshots with styles/preference packs are appreciated.

### Current icon
Classic:
[image]
[image]

Dark contrast:
[image]

Light:
[image]
[image]

### Tabler icon
Classic:
[image]
[image]

Dark contrast:
[image]

Light:
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Reqrefusion, 2024-06-09:
This has nothing to do with the icon. As a general solution, changing the icon according to the text color. Also, the eye suggested by Kadet is obviously better.

NomAnor, 2024-06-09:
> As a general solution, changing the icon according to the text color.

I agree, but Kadets code suffer pixelation on my Setup (4k). This needs to be addressed before chaging it.

> Also, the eye suggested by Kadet is obviously better.

That is a matter of opinion not fact and thats why this issue exists. I'm not against the icon change.

The transparent background inside of the icon makes the lines look to thin in the overlay mode. This could be fixed with a solid background on the inside. Or the generated outline needs to fill it, but this might not be easily possibe.

There are a lot of possible style variants and it might not be possible to support all of them without a lot of work to provide alternative icons by theme.

These setting with Kadets patch
[image]
look like this after a restart:
[image]

In my opinion this does not look good.

I don't know if there is an official list of style settings that should be considered, but at least the three options from the start assistant should look  and work ok, ideally with and without overlay mode.

kadet1090, 2024-06-09:
This is strange - in that case the eye icon should be black as the text is also black - then the icon looks as good as the text does. I'll check why this is the case. As the alternative I can add white fill to the icon so it will still have constant stroke that should help. I however agree that this can be done in another PR.

As for the pixelation - try changing QSize to `(32, 32)` that should help with the pixelation - I'm not able to test with 4k.

NomAnor, 2024-06-09:
It seems that changing the themes in the preferences does not always change the style correctly.
Between changing the style in the preference window and restarting some parts seem to look different, but I have not investigated it. 
It's not only the icon but also the outline of the text. There could be other bugs that are the culprit and not your code.

MisterMakerNL, 2024-06-10:
I have some merge issues in my [darkstyle PR](https://github.com/FreeCAD/FreeCAD/pull/13772#issuecomment-2156042500) which stops me from checking this out.
OVerlay code for the text color has also moved from Qwidget to Qtreeview, so maybe it fixes something here.