FreeCAD/FreeCAD #14567: Exporting an assembly as STEP crashes FreeCAD

prokoudine, 2024-06-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Open an assembly file, e.g. https://github.com/kwahoo2/freecad-motor-driver/blob/main/freecad-assemblies/deltarobot-example/deltarobot-example.FCStd
2. Single-click on the assembly object in the tree, attempt to export as STEP with defaults settings
3. FreeCAD crashes:

```
--------------------------------------------------------------------------------
<string>(1)<class 'AttributeError'>: Module __main__ has no class MotorObserver
<string>(1)<class 'AttributeError'>: Module __main__ has no class MotorObserver
<string>(1)<class 'AttributeError'>: Module __main__ has no class MotorObserver
<Import> ExportOCAF2.cpp(387): deltarobot_example#Joints has null shape
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x796533c45320]
#1  /tmp/.mount_FreeCACb9Zrj/usr/lib/././libTKXDESTEP.so.7.7(+0x89291) [0x7964c7efd291]
#2  /tmp/.mount_FreeCACb9Zrj/usr/lib/././libTKXDESTEP.so.7.7(+0x8a078) [0x7964c7efe078]
#3  0x7964c7eff9e1 in STEPCAFControl_Writer::writeColors(opencascade::handle<XSControl_WorkSession> const&, NCollection_Sequence<TDF_Label> const&) from /tmp/.mount_FreeCACb9Zrj/usr/lib/././libTKXDESTEP.so.7.7+0x9e1
#4  0x7964c7f1cdff in STEPCAFControl_Writer::transfer(STEPControl_Writer&, NCollection_Sequence<TDF_Label> const&, STEPControl_StepModelType, char const*, bool, Message_ProgressRange const&) from /tmp/.mount_FreeCACb9Zrj/usr/lib/././libTKXDESTEP.so.7.7+0x17ef
#5  0x7964c7f1d26f in STEPCAFControl_Writer::Transfer(opencascade::handle<TDocStd_Document> const&, STEPControl_StepModelType, char const*, Message_ProgressRange const&) from /tmp/.mount_FreeCACb9Zrj/usr/lib/././libTKXDESTEP.so.7.7+0x20f
#6  0x7964cefb6ca7 in Import::WriterStep::write(opencascade::handle<TDocStd_Document>) const from /tmp/.mount_FreeCACb9Zrj/usr/lib/./Import.so+0x137
#7  /tmp/.mount_FreeCACb9Zrj/usr/lib/ImportGui.so(+0x32e9d) [0x7964fa31ae9d]
#8  /tmp/.mount_FreeCACb9Zrj/usr/lib/ImportGui.so(+0x2ef6d) [0x7964fa316f6d]
#9  /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libFreeCADBase.so(method_keyword_call_handler+0x374) [0x7965358d4b04]
#10  /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libpython3.11.so.1.0(+0x20603d) [0x79653660603d]
#11  /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7965365e5b5c]
#12  /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7965365f1532]
#13  /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7965366abd1d]
#14  /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7965366ab794]
#15  /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7965366cabf3]
#16  /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x7965366c669a]
#17  /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7965366ba9c9]
#18  0x79653593bdec in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libFreeCADBase.so+0x6c
#19  0x796536f83044 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libFreeCADGui.so+0x94
#20  0x796536e7c8d3 in Gui::Application::exportTo(char const*, char const*, char const*) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libFreeCADGui.so+0x4d3
#21  0x796536f9fcbc in StdCmdExport::activated(int) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libFreeCADGui.so+0x4ac
#22  0x796536f8cf68 in Gui::Command::_invoke(int, bool) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libFreeCADGui.so+0x2a8
#23  0x796536f8d324 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libFreeCADGui.so+0x134
#24  /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7965342eb6f2]
#25  0x79653516a462 in QAction::triggered(bool) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libQt5Widgets.so.5+0x32
#26  0x79653516ce2b in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libQt5Widgets.so.5+0xab
#27  /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libQt5Widgets.so.5(+0x2e12e2) [0x7965352e12e2]
#28  /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libQt5Widgets.so.5(+0x2e8c34) [0x7965352e8c34]
#29  0x7965351aec08 in QWidget::event(QEvent*) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libQt5Widgets.so.5+0x178
#30  0x7965351703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#31  0x796535178262 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libQt5Widgets.so.5+0x13a2
#32  0x796536f38349 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libFreeCADGui.so+0x99
#33  0x7965342b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libQt5Core.so.5+0x108
#34  0x7965351763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#35  /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libQt5Widgets.so.5(+0x1c72a5) [0x7965351c72a5]
#36  /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7965351ca0c9]
#37  0x7965351703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#38  0x796536f38349 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libFreeCADGui.so+0x99
#39  0x7965342b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libQt5Core.so.5+0x108
#40  0x7965349353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libQt5Gui.so.5+0x84c
#41  0x79653490b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libQt5Gui.so.5+0xac
#42  /tmp/.mount_FreeCACb9Zrj/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x79652e794bea]
#43  /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x796532910ad7]
#44  /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x796532913ac7]
#45  /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7965329140ce]
#46  0x79653430aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libQt5Core.so.5+0x66
#47  0x7965342b570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libQt5Core.so.5+0x11b
#48  0x7965342bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libQt5Core.so.5+0x82
#49  0x796536e9b847 in Gui::Application::runApplication() from /tmp/.mount_FreeCACb9Zrj/usr/bin/../lib/libFreeCADGui.so+0xa27
#50  /tmp/.mount_FreeCACb9Zrj/usr/bin/freecad(+0x7620) [0x6145706c9620]
#51  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x796533c2a1ca]
#52  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x796533c2a28b]
#53  /tmp/.mount_FreeCACb9Zrj/usr/bin/freecad(+0x7981) [0x6145706c9981]
```

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37668 (Git) AppImage
Build type: Release
Branch: main
Hash: ea68c5e88cdc80044ede482f6ad908820f4c1c82
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * ExtremeProDark 2.9.0
  * DynamicData 2.60.0
  * Assembly4 0.50.6
  * cura_engine
  * ExplodedAssembly
  * fasteners 0.5.21
  * sheetmetal 0.4.11
  * A2plus 0.4.61
  * BIM021 2021.12.0
  * OpenTheme 2024.4.20
  * Render 2023.12.28
  * Rocket 3.2.4
  * Curves 0.6.21
  * freecad.gears 1.2.0
  * Assembly3 0.12.0
  * boltsfc 2022.11.5
  * OpenDark 2023.12.17
  * dodo 1.0.0
  * Ondsel-Lens 2024.4.16.01
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

prokoudine, 2024-06-09:
@PaddleStroke Is this up your alley?

maxwxyz, 2024-06-11:
I cannot reproduce on 
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37669 (Git)
Build type: Release
Branch: main
Hash: 3b4598ce04d7aaec69feeb560c29ae18f0236e0c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CurvedShapes 1.0.9
  * Curves 0.6.36
  * fasteners 0.5.21
  * OpenTheme 2024.5.16
  * OpticsWorkbench 1.0.17
  * sheetmetal 0.4.14

```

PaddleStroke, 2024-06-11:
Cannot reproduce either on win10

maxwxyz, 2024-06-11:
@prokoudine could you provide an example file where this is reproducible? Maybe one body is broken.

prokoudine, 2024-06-11:
Er, I already did provide?

maxwxyz, 2024-06-11:
ah sorry, wrong issue. As I said I cannot reproduce this unfortunately.

prokoudine, 2024-06-11:
It seems to be Linux-specific. Another Linux user reported this on Ondsel's Discord.

PaddleStroke, 2024-06-12:
Can the same bug be reproduced with a App::Part instead of an assembly?

liamhays, 2024-06-15:
I can confirm this issue on Linux with that specific file, once you download all the other files that are part of the assembly. However, loading the assembly also causes this sequence of errors in report view (visible in OP's report):

```
13:14:41  pyException: <string>(2)<class 'AttributeError'>: Module __main__ has no class MotorObserver
13:14:41  pyException: <string>(2)<class 'AttributeError'>: Module __main__ has no class MotorObserver
13:14:41  pyException: <string>(2)<class 'AttributeError'>: Module __main__ has no class MotorObserver
```
and I wonder if those are causing the problem. Specs:
```
OS: Arch Linux (KDE)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git) AppImage
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

On the actual export, this is printed on both systems (could also be the error source, especially if it's caused by broken bodies):
```
<Import> ExportOCAF2.cpp(387): deltarobot_example#Joints has null shape
```
and then a segmentation fault occurs, crashing the application. On Windows (specs below), a segfault also occurs, but something catches it and makes a little dialog that says "Export failed: Access violation". Clicking OK closes the dialog and FreeCAD doesn't crash. Windows specs:

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37730 (Git)
Build type: Release
Branch: main
Hash: 30f7f2688921e6e5242beaa9d71e0fdd844b20e3
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

prokoudine, 2024-06-15:
> and I wonder if those are causing the problem

Most unlikely, because the original report came from a user who wasn't using MotorObserver.

luzpaz, 2024-07-22:
Should this be tagged as 'Blocker' ?

prokoudine, 2024-08-06:
I'd say yes.

luzpaz, 2024-08-06:
Tagged as Blocker but of course defer to the devs to make the decision

bgbsww, 2024-08-17:
I do see the missing MotorObserver errors on loading, but I don't have any problems or crashes when exporting the STEP file, including after I repair the missing links.  So another can't reproduce :(

chennes, 2024-08-18:
I can reproduce on my MacOS system (I see the `<Import> ExportOCAF2.cpp(387): deltarobot_example#Joints has null shape` on the command line after the segfault):
```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38480 (Git)
Build type: Release
Branch: addonManagerZipInstallationSubfolderIssue
Hash: 20549cacd2d61aec16b6b7873fb73962a6241d3b
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

chennes, 2024-08-18:
Backtrace:
```
<unknown> 0x0000000153e50190
<unknown> 0x0000000153e46104
STEPCAFControl_Writer::writeColors(opencascade::handle<XSControl_WorkSession> const&, NCollection_Sequence<TDF_Label> const&) 0x0000000153e3b850
STEPCAFControl_Writer::transfer(STEPControl_Writer&, NCollection_Sequence<TDF_Label> const&, STEPControl_StepModelType, char const*, bool, Message_ProgressRange const&) 0x0000000153e395d4
STEPCAFControl_Writer::Transfer(opencascade::handle<TDocStd_Document> const&, StepData_ConfParameters const&, STEPControl_StepModelType, char const*, Message_ProgressRange const&) 0x0000000153e38920
STEPCAFControl_Writer::Transfer(opencascade::handle<TDocStd_Document> const&, STEPControl_StepModelType, char const*, Message_ProgressRange const&) 0x0000000153e38728
Import::WriterStep::write(opencascade::handle<…>) const WriterStep.cpp:51
ImportGui::Module::exporter(const Py::Tuple &, const Py::Dict &) AppImportGuiPy.cpp:540
Py::ExtensionModule::invoke_method_keyword(void *, const Py::Tuple &, const Py::Dict &) ExtensionModule.hxx:192
method_keyword_call_handler(PyObject *, PyObject *, PyObject *) cxx_extensions.cxx:1693
cfunction_call 0x0000000101aa2d88
_PyObject_MakeTpCall 0x0000000101a51348
_PyEval_EvalFrameDefault 0x0000000101b45580
PyEval_EvalCode 0x0000000101b3a730
run_eval_code_obj 0x0000000101b9bb3c
run_mod 0x0000000101b99bd4
PyRun_StringFlags 0x0000000101b992b8
Base::InterpreterSingleton::runString(const char *) Interpreter.cpp:249
Gui::Command::_runCommand(const char *, int, Gui::Command::DoCmd_Type, const char *) Command.cpp:725
Gui::Application::exportTo(const char *, const char *, const char *) Application.cpp:785
StdCmdExport::activated(int) CommandDoc.cpp:486
Gui::Command::_invoke(int, bool) Command.cpp:444
Gui::Command::invoke(int, Gui::Command::TriggerSource) Command.cpp:408
Gui::Action::onActivated() Action.cpp:111
QtPrivate::FunctorCall::call(void (Gui::Action::*)(), Gui::Action *, void **) qobjectdefs_impl.h:145
QtPrivate::FunctionPointer::call<…>(void (Gui::Action::*)(), Gui::Action *, void **) qobjectdefs_impl.h:182
QtPrivate::QCallableObject::impl(int, QtPrivate::QSlotObjectBase *, QObject *, void **, bool *) qobjectdefs_impl.h:553
void doActivate<false>(QObject*, int, void**) 0x0000000109339ca8
QAction::activate(QAction::ActionEvent) 0x0000000108e95008
QObject::event(QEvent*) 0x00000001093347dc
QApplicationPrivate::notify_helper(QObject*, QEvent*) 0x00000001085487f4
QApplication::notify(QObject*, QEvent*) 0x000000010854965c
Gui::GUIApplication::notify(QObject *, QEvent *) GuiApplication.cpp:83
QCoreApplication::notifyInternal2(QObject*, QEvent*) 0x00000001092fe0ec
QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) 0x00000001092fee7c
QCocoaEventDispatcherPrivate::processPostedEvents() 0x000000010a8ca168
QCocoaEventDispatcherPrivate::postedEventsSourceCallback(void*) 0x000000010a8cae18
__CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ 0x00000001847ce4d8
__CFRunLoopDoSource0 0x00000001847ce46c
__CFRunLoopDoSources0 0x00000001847ce1dc
__CFRunLoopRun 0x00000001847ccdc8
CFRunLoopRunSpecific 0x00000001847cc434
RunCurrentEventLoopInMode 0x000000018ef7019c
ReceiveNextEventCommon 0x000000018ef6fe2c
_BlockUntilNextEventMatchingListInModeWithFilter 0x000000018ef6fd30
_DPSNextEvent 0x000000018802bd68
-[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] 0x0000000188821808
-[NSApplication run] 0x000000018801f09c
QCocoaEventDispatcher::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) 0x000000010a8c927c
QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) 0x00000001093058a0
QCoreApplication::exec() 0x00000001092fe620
tryRunEventLoop(Gui::GUISingleApplication &) Application.cpp:2062
runEventLoop(Gui::GUISingleApplication &) Application.cpp:2082
Gui::Application::runApplication() Application.cpp:2159
main MainGui.cpp:300
start 0x00000001843660e0
```

chennes, 2024-08-18:
@bgbsww (or anyone else) are you able to do a recompute on this file? I get several errors when I do so, including an over constrained sketch.

bgbsww, 2024-08-18:
Correct, I had to fix all the recompute errors in the file in order to
test.  There's a missing object referred to, and the overconstrained sketch.

chennes, 2024-08-19:
Interesting, I just followed Alex's steps, without the recompute. So presumably the crash is related to something being unable to handle the bad geometry, and behaving in the worst way possible.

bgbsww, 2024-08-19:
I did that too, and didn't get the crash, which is why I then tried the other steps.  I agree with your assessment.

CalligaroV, 2024-08-29:
Hi Everyone.

I was looking at this issue and noticed a couple of things:
- if I try to export the assembly with the option `Export invisible objects` **unchecked** I can export it without issues, even after a recompute
- I can export the assembly also with the option `Export invisible objects` **checked** but before I have to delete all the hidden bodies (for example `arm-skeleton` inside the part `arm`)

@prokoudine can you do a similar test and confirm if my findings can be reproduced also by your side?

I'm able to reproduce this issue also with this example file I made for tests:
[PartWithHiddenBody.FCStd.zip](https://github.com/user-attachments/files/16804267/PartWithHiddenBody.FCStd.zip)

```
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Debug
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Installed mods: 
  * Assembly4 0.50.15
  * freecad.gears 1.2.0
  * Beltrami 1.1.0
  * sheetmetal 0.4.24
  * Curves 0.6.44
  * fasteners 0.5.26
  * kicadStepUpMod 11.2.7
  * OpenTheme 2024.8.10
  * Render 2024.8.4
  * Ondsel-Lens 2024.7.5.02
```

chennes, 2024-08-31:
By definition this can't be a regression (since it requires an Assembly), and it appears to require broken geometry to trigger. I propose removing the blocker tag: making the STEP export recognize the incoming failing geometry is really probably a feature request more than a blocking bug. @maxwxyz what do you think?

maxwxyz, 2024-08-31:
done

luzpaz, 2024-09-19:
I don't think we can tag this as a FR until we fix the crash ?

picli3, 2024-09-23:
> Hola a todos.
> 
> Estaba analizando este problema y noté un par de cosas:
> 
> * Si intento exportar el ensamblaje con la opción `Export invisible objects` **desmarcada,** puedo exportarlo sin problemas, incluso después de un recálculo.
> * También puedo exportar el ensamblaje con la opción `Export invisible objects` **marcada** , pero antes tengo que eliminar todos los cuerpos ocultos (por ejemplo, `arm-skeleton`dentro de la pieza `arm`)
> 
> @prokoudine¿Puedes hacer una prueba similar y confirmar si mis hallazgos pueden ser reproducidos también por tu parte?
> 
> También puedo reproducir este problema con este archivo de ejemplo que hice para las pruebas: [PartWithHiddenBody.FCStd.zip](https://github.com/user-attachments/files/16804267/PartWithHiddenBody.FCStd.zip)

This worked for me, I was also having the same problem when exporting an assembly.
```
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git) AppImage
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * A2plus 0.4.67
  * kicadStepUpMod 11.3.8
  * Behave-Dark-Colors 0.1.1
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.4.26
```

prokoudine, 2024-09-23:
> This worked for me

As in "bug reproduced" or the opposite of that? :)

picli3, 2024-09-23:
> > Esto me funcionó
> 
> ¿Como en "error reproducido" o lo opuesto de eso? :)

as a solution to the error. :D

PaddleStroke, 2025-07-18:
The crash happens in ExportOCAF2::exportObject

Here are some more information and details about various cases where I put some logs in the exportObject
First problematic but not crashing : 
Part
- hiddenbody
- hiddenbody2

result :
```
10:29:15  exportObject PartWithHiddenBody#Part
10:29:15  <Import> ExportOCAF2.cpp(393): PartWithHiddenBody#Part has null shape
```

Second case: wOK
Part
- visiblebody
- visiblebody2

Result : 
```
10:30:19  exportObject PartWithHiddenBody#Part
10:30:19  1 obj PartWithHiddenBody#Part
10:30:19  2 PartWithHiddenBody#Part
10:30:19  exportObject PartWithHiddenBody#Part
10:30:19  1 obj PartWithHiddenBody#Body
10:30:19  2 subs empty PartWithHiddenBody#Part
10:30:19  2 end
10:30:19  exportObject PartWithHiddenBody#Part
10:30:19  1 obj PartWithHiddenBody#Body001
10:30:19  2 subs empty PartWithHiddenBody#Part
10:30:19  2 end
10:30:19  3 PartWithHiddenBody#Part
10:30:19  4 PartWithHiddenBody#Part
10:30:19  end PartWithHiddenBody#Part
```

The most problematic case : 
Part
- hiddenbody
- visiblebody

```
08:54:50  exportObject PartWithHiddenBody#Part
08:54:50  1 obj PartWithHiddenBody#Part
08:54:50  2 PartWithHiddenBody#Part
08:54:50  exportObject PartWithHiddenBody#Part
08:54:50  1 obj PartWithHiddenBody#Body
08:54:50  2 subs empty PartWithHiddenBody#Part
08:54:50  2 end
08:54:50  exportObject PartWithHiddenBody#Part
08:54:50  1 obj PartWithHiddenBody#Body001
08:54:50  2 subs empty PartWithHiddenBody#Part
08:54:50  2 end
08:54:50  3 PartWithHiddenBody#Part
08:54:50  4 PartWithHiddenBody#Part
08:54:50  end PartWithHiddenBody#Part
08:54:50  Illegal storage access...
08:54:50  pyException: Traceback (most recent call last):
  File "<string>", line 6, in <module>
<class 'OSError'>: Illegal storage access! Please save your work under a new file name and restart the application!
```

What does that tell us?

For a container object like App::Part, the Part::Feature::getTopoShape method, by default, computes a compound shape consisting of its visible children only. But with the exportHidden option, we are still including the hidden childs. So there's a mistmatch between the shape and the OCAF assembly we are making. Which at some point later ends up in the crash.

PaddleStroke, 2025-07-18:
Ok with the new flag now the shape is generated correctly. But it still crashes.
I think it's because of the OCCT bug mentioned. Let's see if it can be fixed

PaddleStroke, 2025-07-18:
YES finally found it.

In `ExportOCAF2::exportObject` there this block : 
```
        if (!vis) {
            // Work around OCCT bug. If no color setting here, it will crash.
            // ...
            auto childShape = aShapeTool->GetShape(childLabel);
            Quantity_ColorRGBA col;
            if (!aColorTool->GetInstanceColor(childShape, XCAFDoc_ColorGen, col)
                && !aColorTool->GetInstanceColor(childShape, XCAFDoc_ColorSurf, col)
                && !aColorTool->GetInstanceColor(childShape, XCAFDoc_ColorCurv, col)) {
                auto& c = options.defaultColor;
                aColorTool->SetColor(childLabel, Tools::convertColor(c), XCAFDoc_ColorGen);
                FC_WARN(Tools::labelName(childLabel) << " set default color");
            }
            aColorTool->SetVisibility(childLabel, Standard_False);
        }
```
The problem is that the if condition is never true. And so the fix is never applied.

Removing this condition and then it doesn't crash anymore. Now the issue is that we are loosing the color information I guess. I'll check if I can find a better fix.

PaddleStroke, 2025-07-18:
Note currently we don't have a 'ShapeColor' property anymore. But in 0.21 we still had this property.
Perhaps it's this removal that is actually causing the crash.

closed 2025-07-18 by chennes