FreeCAD/FreeCAD #14568: Transparent objects are not rendered transparent

NomAnor, 2024-06-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Changing the Transparency property with the toggle or by hand does not render the object transparent.

[image]

Hovering over the tree item renders the object transparent.
[image]

These screenshots are with the default settings from an empty user and system settings file.

### Full version info

```shell
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37330 (Git)
Build type: Debug
Branch: main
Hash: a094bf0df7e2b3e5c5b2e3b158a9040a6474892f
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.3.0
Locale: German/Germany (de_DE)
Installed mods: 
  * Manipulator 1.5.7
  * ThreadProfile 1.84.0
  * DynamicData 2.62.0
  * Assembly3.backup1715897154.0970206 0.11.3 (Disabled)
  * lattice2 1.0.0
  * 3D_Printing_Tools
  * freecad.gears 1.2.0
  * fasteners 0.4.21
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

leoheck, 2024-06-09:
I am also experiencing this...

The console shows that the "Toggle transparency" always shows the same msg.
[image]

Also, the shortcut does not always work... something it does not produce anything on the console, and sometimes it does.

```
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37668 (Git) AppImage
Build type: Release
Branch: main
Hash: ea68c5e88cdc80044ede482f6ad908820f4c1c82
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * POV-Ray-Rendering
  * Assembly4 0.50.12
  * sheetmetal 0.4.17
  * InventorLoader 1.4.0
  * ose-piping
  * fasteners 0.5.21
  * asm4_disable_version_check_popup
  * OpenDark 2024.3.13
  * ProDarkThemePreferencePack 1.0.0
  * Render 2024.6.5
  * OpticsWorkbench 1.0.17
  * lattice2 1.0.0
  * kicadStepUpMod
```

lmutt, 2024-06-10:
Also found this bug here after upgrading from 0.22.0dev.37573 to 0.22.0dev.37668.

```
OS: Arch Linux (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37668 (Git) AppImage
Build type: Release
Branch: main
Hash: ea68c5e88cdc80044ede482f6ad908820f4c1c82
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Brazil (pt_BR)
Installed mods: 
  * fasteners 0.5.22
  * QuickMeasure 2022.10.28
  * ThreadProfile 1.89.0
  * Manipulator 1.5.7
  * OpenTheme 2024.5.16

```

lmutt, 2024-06-11:
**Update**: 0.22.0dev.37669 (just released) also has this bug.

**More details**: "Toggle transparency" from menu (or the shortcut V,T) don't change nothing. Changing the "transparency" property ("View" tab from property view) also do nothing.

BUT, if after Toggle transparency I choose "Random color" command from menu, it apply the last Toggle transparency command. Also changing the "Shape Appearance" color (again View tab from property view) make the Toggle transparency apply.

```
OS: Arch Linux (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37669 (Git) AppImage
Build type: Release
Branch: main
Hash: 3b4598ce04d7aaec69feeb560c29ae18f0236e0c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Portuguese/Brazil (pt_BR)
Installed mods: 
  * fasteners 0.5.22
  * QuickMeasure 2022.10.28
  * ThreadProfile 1.89.0
  * Manipulator 1.5.7
  * OpenTheme 2024.5.16

```

maxwxyz, 2024-06-11:
@PaddleStroke @davesrocketshop FYI

davesrocketshop, 2024-06-11:
A previous bug meant that the transparency was being set every time. When that bug was fixed it created conditions where the transparency property value was set, but not rendered.

The toggle transparency command will always show the same message. It does not reflect the current or desired transparency setting.

leoheck, 2024-06-13:
Transparency is the most important feature. Any chance to get this fixed ASAP?

Also, how do you use this Material thing? This area is so small I cannot even understand what it shows.
I can see only 2 lines and this is kind of not useful.
[image]

davesrocketshop, 2024-06-14:
The fix is in, just waiting for the merge. I don't control that process. It should be merged after the Monday PR review.

As for the dialog size, I agree. I don't have full control over how the task dialogs are handled, for example there is no resize option, but I'll see if there are any ways to improve it.

leoheck, 2024-06-14:
If there is no control over the how it will look, then it should be replaced by something else like a button that launches a pop-up or something, I believe.

davesrocketshop, 2024-06-15:
As it turns out, I hadn't set any minimal size for the widget. This will be an easy fix. I have created a seperate issue #14686 to track this as it is unrelated to the original issue. They should become available at about the same time

PR #14615 (2024-06-17): Materials: Render Transparency
Under certain conditions an objects transparency would no be updated.

fixes #14568
commit 417ca729 (2024-06-17): Materials: Render Transparency

Under certain conditions an objects transparency would no be updated.

fixes #14568
closed 2024-06-17 by chennes