FreeCAD/FreeCAD #14569: Building from latest main branch, still seeing what seems to be a TopoNaming issue

dresco, 2024-06-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Am building from latest main, and able to replicate an issue that I'd hoped would go away with the toponaming fixes. (I understand these fixes should be enabled in current builds from source?).

Attaching a simplified test file. This file contains two sketches, one containing reference geometry, and one that imports external geometry (points) from the reference sketch, and simply creates vertical lines between them. 

As originally created, the top and bottom points on the **left side** of each upright section were selected using the create external geometry tool;

[image]

However when the dimensions of the reference sketch are changed sufficiently, the external geometry has changed to the **right side** of some of the upright sections;

[image]

[Test.zip](https://github.com/user-attachments/files/15752983/Test.zip)

Was hoping the second sketch would stay consistent with the toponaming fixes, but perhaps I'm missing something?

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37669 (Git)
Build type: Unknown
Branch: main
Hash: 3b4598ce04d7aaec69feeb560c29ae18f0236e0c
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * OpenTheme 2024.5.16
  * FeedsAndSpeeds 0.5.0
  * kicadStepUpMod 11.1.5
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-11:
@bgbsww is this a duplicate of https://github.com/FreeCAD/FreeCAD/issues/14563 ?

bgbsww, 2024-06-15:
Probably due to same cause as #14563, we'll see, I'm looking now.

bgbsww, 2024-06-17:
I was wrong.  This is not the same as #14563, and it tellingly has a series of warnings in the console:
`11:36:35  <PropertyLinks> PropertyLinks.cpp(453): Test#Sketch001.ExternalGeometry missing element reference Test#Sketch ;g8v1;SKT;:H,V.Vertex5`

Those make it clear that the external geometry lookups are failing, for reasons yet to be determined, but we'll look.

dresco, 2024-06-17:
Thanks, don't know if it will help or hinder, but I created that very stripped down file to demonstrate an issue I was having with my real file. Attaching the real one here for reference also, just in case the more complex case helps...

In this case, most of the walls are modelled with a reference sketch, and then a series of sketches that are used as the paths for Arch -> Frame extrusions. Same concept of picking external geometry start/end points from the reference sketch for each path.

I also see all these external geometry lookups failing whenever a reference sketch is updated or sometimes even opened & closed.. (Not sure why it sometimes happens without even making any changes to the sketch, possibly related to opening the file with a newer build than it was saved with?)

Example steps to reproduce (this file fails consistently for me now, using latest code built today);
Open right_wall_stud_sketch, will see that it's fully constrained.
Close right_wall_stud_sketch.
Open right_wall_reference_sketch, will see that it's also fully constrained.
Close right_wall_reference_sketch.
Will see the errors about invalid shape names pop-up.
Re-open right_wall_stud_sketch, will be missing all it's constraints.

[Garden Building v5.zip](https://github.com/user-attachments/files/15874556/Garden.Building.v5.zip)

Hope that's of some use, thanks!

bgbsww, 2024-06-17:
Thanks.  The first example is great and is showing where a lookup is failing.  Unfortunately it is failing because it is looking in a map that wasn't populated, and that's a little more complicated to sort out, but I'm on the path,

bgbsww, 2024-07-06:
Using the latest code, the original test file seems to behave correctly - no more missing references showing up, and I could change all the constraints without any toggling - except when I set the overall width narrow enough that it caused the right-most joist to toggle due to normal sketcher solver behavior.  Still without the missing references, which was the toponaming problem.

However, the Garden_Building_v5 is interesting in that ?Vertex1 through ?Vertex4 were all saved as external references from Sketch035 (roof_joist_sketch_002) to Sketch034 (roof_joist_sketch ) meaning that the references were broken when the file was saved.  That file is causing crashes for me if I look at it sideways, but it also seems deeply broken.  I'm going to keep looking at it.

It particular, many of the sketches need recomputation, and  when recomputed lead to segfaults trying to ElementMap::addName.

This seems relevant:

```
<class 'AttributeError'>: Property container has no property 'PlacementList'
Traceback (most recent call last):
  File "/home/brad/git/FreeCAD/cmake-build-debug/Mod/Draft/draftobjects/array.py", line 67, in onDocumentRestored
    super(Array, self).onDocumentRestored(obj)
  File "/home/brad/git/FreeCAD/cmake-build-debug/Mod/Draft/draftobjects/draftlink.py", line 174, in onDocumentRestored
    self.linkSetup(obj)
  File "/home/brad/git/FreeCAD/cmake-build-debug/Mod/Draft/draftobjects/array.py", line 343, in linkSetup
    super(Array, self).linkSetup(obj)
  File "/home/brad/git/FreeCAD/cmake-build-debug/Mod/Draft/draftobjects/draftlink.py", line 126, in linkSetup
    obj.setPropertyStatus('PlacementList', '-Immutable')
<class 'AttributeError'>: Property container has no property 'PlacementList'

```

dresco, 2024-07-07:
Thank you.

If it helps, I can make a copy of the file, remove all the existing external geometry and then re-add it (using the latest code). See whether I can get back to a clean state and whether it happens again?

luzpaz, 2024-07-18:
Can we rename this issue to be less generic ? What exactly is triggering TNP here ?

bgbsww, 2024-07-18:
Files with saved missing element references can crash the app.

There is no longer a TNP, just a robustness problem in that files that had detected missing references may not load cleanly.  Not sure if this is only for files from broken development builds, or if it applies generally.

I expect this to either go away as a side effect of other changes being make to the load/save code - there was a lot of missing stuff, or to be classified as no-longer repeatable, but the file here is useful to have to prove that.

maxwxyz, 2024-07-19:
I have removed the blocker tag then.

luzpaz, 2024-07-21:
Renamed per https://github.com/FreeCAD/FreeCAD/issues/14569#issuecomment-2237647561

bgbsww, 2024-07-30:
This needs a retest to see if anything is still there.  I don't think you can make files with missing element references anymore.

maxwxyz, 2024-07-30:
@dresco could you please report back?

dresco, 2024-07-31:
> @dresco could you please report back?

Thanks yes, I will remove & re-add the external geometry links and see whether it's still fragile.
Please give me a day or two..

dresco, 2024-08-05:
Can confirm my test file is behaving much better with a current dev build. I've recreated all the external geometry references, and haven't experienced any problems since.

@bgbsww Thanks very much for all your efforts! Happy to close this issue...

closed 2024-08-05 by dresco