FreeCAD/FreeCAD #14570: Mouse pointer disappearing on application focus

Farmer-Al, 2024-06-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am running Ubuntu Ubuntu 22.04.4 LTS an with the FreeCAD 0.21.2 release. 

Whenever I give the application application focus, my mouse pointer disappears inside the FreeCAD application.   If I right click, it will show up again, until I click, then it is hidden again.  This happens ONLY when the application is full screen or maximized.

### Full version info

```shell
[code]
OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) Snap 908
Build type: Release
Branch: tag: 0.21.2
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

HumbleDeer, 2024-06-09:
Does it disappear entirely, or does a different cursor appear?

Farmer-Al, 2024-06-09:
Disappears entirely.  The only indication I still have a pointer is things highlight as though the pointer is there.

xtemp09, 2024-06-10:
Mouse pointers are supposed to be hardware-accelerated, so this problem comes from drivers or from _Gnome_.

The solution may be to use software mouse cursor or Gnome tweaks.

HumbleDeer, 2024-06-10:
Addendum to this then, check your drivers!

Farmer-Al, 2024-06-10:
Confirmed, another application, when maximized, is loosing the cursor as well.   Sorry for the fire drill, and thank you for the help narrowing it down.

On a side note, great work on the software though!

HumbleDeer, 2024-06-10:
Good to hear your issue has been identified.

Since this doesn't seem related to freecad anymore, you might want to close
this issue.