FreeCAD/FreeCAD #14571: Vertical constraint display problem.

Malendryn, 2024-06-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

See attached file and 'Sketch341' inside.  Open the sketch and select 'Constraint 15 in the Constraints window.

At first you can barely see/find the constraint, however if you now scrollwheel zoom in closer and closer you can see the constraint move somewhat haphazardly right until you zoom in fifteen times, and then it 'snaps' into its correct position.

If you zoom back out, it moves again, hovering just to the left of constraint 33 in the visual, but after about four more zoomouts it suddenly jumps significantly again.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.5.15
  * freecad.gears 1.2.0
```

### Subproject(s) affected?

None

### Anything else?

[Test.zip](https://github.com/user-attachments/files/15753127/Test.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Malendryn, 2024-06-09:
Note, the constraints on this are also in-conflict somewhat, but the solver didn't report them as such.  I was attempting to 'verticalize' the two rightmost legs just below the arcs but it was failing, yet it did accept the constraints into the Constraints window, and didn't complain about them. (however it also didn't verticalize the lines either)

maxwxyz, 2024-06-11:
@FlachyJoe FYI

FlachyJoe, 2024-06-11:
Pictures of the "problem"

[image]

[image]

[image]

In order the icons do not overlap they are listed together. It's a design choice so we can roughly see where the concerned elements are.

maxwxyz, 2024-06-11:
ah, ok then it's not bug but intended so you can see all constraints.

closed 2024-06-11 by maxwxyz