FreeCAD/FreeCAD #14573: Path dressups adds unneeded line causing CNC going back to origin

Team1Electronics, 2024-06-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

**Describe the problem and how it impacts user experience, workflow, maintainability or speed of the code:**
When using dressups (eg ramp entry) when performimg multiple operations, the dressup causes an extra line to be added to the G-code causing the machine to go to origin at the start of each operation. 

This extra line is added before each operation that uses a dressup: 
`G0 F0.000000 X0.000000 Y0.000000 Z5.000000`

**If the problem appears to be a bug with the current functionality, provide as test case or recipe that reproduces the error.**
There is a post in the freecad forums that describes in detail how to reproduce this problem and also confirms it is indeed the dressup feature and not eg the postprocessor:

[https://forum.freecad.org/viewtopic.php?p=764348#p764348](url)

**Short problem description**
When using multiple profile operations, freecad shows the machine to go to the origin bafore eacht operation, but when posting the machine does NOT go to the origin as indicated in freecad. The problem only happens when you use a dressup (eg ramp entry) on the profile. Because of the dressup, the toolpaths don't only appear to be going to the origin, but it's also posted to the NC file.

**Example**
using a profile operation for some pockets, you will see in freecad the machine goes to the origin between each operation. In the posted file (visualized in linuxcnc here) the machine does not actually go to the origin -> OK, no problem:

No dressups used - Freecad shows the machine going back to the origin:
[image]

But the posted file does not cause the CNC to actually go to the origin before the start of each profile:
[image]

Same file, but using dressups:
[image]

And when posting, an extra line is added before each operation causing the CNC machine to actually go to the origin which is not wanted. **This is the problem.**
[image]

### Full version info

```shell
tested in 2 version:

[code]
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.12
  * fasteners 0.5.21
  * A2plus 0.4.64a
  * Assembly3 0.12.2
  * QuickMeasure 2022.10.28
  * kicadStepUpMod 11.1.5
  * btl 0.9.9
  * FeedsAndSpeeds 0.5.0
[/code]

[code]
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.12
  * fasteners 0.5.21
  * A2plus 0.4.64a
  * Assembly3 0.12.2
  * QuickMeasure 2022.10.28
  * kicadStepUpMod 11.1.5
  * btl 0.9.9
  * FeedsAndSpeeds 0.5.0
[/code]
```

### Subproject(s) affected?

None

### Anything else?

Post describing the problem in detail with steps to reproduce:

https://forum.freecad.org/viewtopic.php?p=764348#p764348

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Team1Electronics, 2024-06-09:
Problem also present in MAC version as mentioned by csrichardson

OS: macOS 12.7.4
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * SelectorToolbar
  * Manipulator 1.5.7
  * CurvedShapes 1.0.9
  * Help 1.0.3
  * Curves 0.6.37

maxwxyz, 2024-06-11:
related to https://github.com/FreeCAD/FreeCAD/issues/12459 ?

Team1Electronics, 2024-06-11:
> related to #12459 ?

Don't think so. #12459 seems to be related to a calculation problem in the tag dressup. Issue #14573 is about an extra G0 command being inserted when using the profile dressup. But #12459 might need higher priority as that would indeed completely block people from using that dressup...

Issue #12459:
[image]

luzpaz, 2024-08-27:
@Team1Electronics can you reproduce on latest development build ?
But since you're on macOS, I don't recommend downloading the dev build till is  #293 fixed (it shows merged ATM but there are still issues).

Team1Electronics, 2024-08-27:
> @Team1Electronics can you reproduce on latest development build ? But since you're on macOS, I don't recommend downloading the dev build till is #293 fixed (it shows merged ATM but there are still issues).

Yes of course! Will test this as soon as I get back home and report back. Thanks for taking a look at this!!
Ps - I'm not on macOS but on Linux or windows :-)

JeffT65, 2024-09-06:
I have been using for a few months and have had the same issue. I am on windows. I downloaded Freecad 0.22.0dev but still have the same issue. Only happens after adding ramp dressup (all 3 ramp types). Does not happen on tag or LeadInOut dressups. 
[image]

MikeyC3040, 2024-09-29:
This was also driving me a bit crazy. I've made a modification to RampEntry.py that so far is working for me. Simply I just ignore adding the x/y to the command if the path edge shows no x and no y changes.
[https://github.com/MikeyC3040/FreeCAD/blob/RampEntryZero/src/Mod/CAM/Path/Dressup/Gui/RampEntry.py](url)

luzpaz, 2024-09-29:
Any CAM folks willing to take a look at this ?

MikeyC3040, 2024-09-29:
[https://github.com/MikeyC3040/FreeCAD/commit/6ad29cf7a8e68e8a28902cba3c8bce2dffea6cd0](url)

luzpaz, 2024-09-29:
@maxwxyz any CAM folks we can defer to to review this ?

maxwxyz, 2024-09-30:
@sliptonic @luvtofish maybe?

J-Dunn, 2025-07-14:
I fixed what looks like a similar bug a few years ago since it kept smashing my tools into the clamps. 

IIRC that was probably on boundary dressup. Not sure how I could find that again now. 
Sounds like a similar logical error in the algo. Sliptonic may recall some interaction on that at the time ( if he has better memory this I do).

tarman3, 2025-07-14:
Already fixed by #21519

J-Dunn, 2025-07-14:
Thanks for the fix tarman.  It indeed looks like the same bug in ramp dressup as I found in boundary.

Your merged PR is nice and short but what if the initial position really is 0,0 which is legit and I would think quite possible, I often set the corner of my stock to 0,0,0 . 

Maybe we should consider possible bad effects of incorrectly removing the first move as well. 

I have not opened the code but I'd guess the problem is the same : a bogus edge gets created from 0,0 to the true starting point of the path. That only works correctly when the true starting point is 0,0 in which case the fake edge move is a NOP.  Your fix simple removes that. 

It seems the fix inverts the bug rather than cleanly fixing the problem. 
 
It seems I should have checked all the other dressups at the time, having found a fix. I probably did not have time.

J-Dunn, 2025-07-14:
Profile only: 

```
M3 S7500
(Finish operation: TC: 10mm_Endmill001)
(Begin operation: Profile)
(Path: Profile)
(Profile)
(Compensated Tool Path. Diameter: 5.0)
G0 Z6.000
G0 X16.515 Y16.515
G0 Z4.000
G1 X16.515 Y16.515 Z-4.000 F180.000

```
Ramp dressup:
```
(M6 T20)
M3 S7500
(Finish operation: TC: 10mm_Endmill001)
(Begin operation: RampEntryDressup)
(Path: RampEntryDressup)
G0 X0.000 Y0.000 Z6.000
G0 X16.515 Y16.515 Z6.000
G0 X16.515 Y16.515 Z4.000
```

RiccaDS, 2025-08-10:
experiencing the same problem here in dev 42885

jffmichi, 2026-03-20:
#22484 is merged now. I think this could be closed.