FreeCAD/FreeCAD #14575: Improve linking new parts to an assembly

prokoudine, 2024-06-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This request comes from a user on Discord. As it covers general assembly functionality, I think this belongs to the upstream tracker. Here's the quote:

> My question is whether Ondsel already has a functionality to copy and paste like this? (With the CTRL key, drag the object and position it immediately where I want) What Ondsel does also does it, but it takes much longer, is not so intuitive, and the objects all end up on top of each other if they are all the same. For example, for screws, it's very handy. 
And CTRL C + CTRL V pastes it where the mouse is positioned

https://github.com/FreeCAD/FreeCAD/assets/57467/b1e8f631-61ad-43c8-80bf-b5b47601e5b7

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of Ondsel: 64-bit
Version: 2024.2.1.37226 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 2024.2.1)
Hash: 1a1bd7290494aef925a026c87b808ac58d53c3b0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.5.21
  * slic3r-tools
  * OpenTheme 2024.4.20
  * Lattice2 1.0.0
  * Curves 0.6.37
  * freecad.gears 1.2.0
  * Ondsel-Lens 2024.5.6.01
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-11:
I get the intended feature request but what is the underlying problem to be solved? Maybe there is a better solution.

maxwxyz, 2024-06-11:
Would want to add to this: First part is inserted at random location, not with their origin to the assembly origin, so I manually have to reset the position and then ground it.

luisDias89, 2024-06-17:
@maxwxyz I was reading this, and I totally agree with this problem.
This is a matter of productivity and simplicity.
If you have a big assembly with 10,000 parts and you want to place 10 screws in a specific area of the design, this helps preposition each one in the right place.
What you can do in 30 seconds in SolidWorks can take 10 minutes the way it is in FreeCAD.

luisDias89, 2024-06-17:
Try to do the same as in the video, but in FreeCAD to see how long it takes you.

luisDias89, 2024-06-17:
> @maxwxyz I was reading this, and I totally agree with this problem. This is a matter of productivity and simplicity. If you have a big assembly with 10,000 parts and you want to place 10 screws in a [...]

PS: And I'm not even talking about automatically detecting that the screw can be inserted and automatically creating a joint.

PaddleStroke, 2024-07-04:
The insert component tool was initially kind of working like in the video. You would click on an item, then move the mouse over the 3d view and the object would follow.
It was then modified to show the dragger instead.

But now that I think on it, we can also define a behavior for click and drag. So if you click and drag then it would follow the mouse.