FreeCAD/FreeCAD #14577: BaseFeature doesn't have TNP mitigations

Jookia, 2024-06-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Do the following things:

1. Create two cubes
2. Shift the second cube so a quarter of it intersects the first
3. Fusion them
4. Create a PartDesign body
5. Drag the fusion in to PartDesign
6. Place a sketch on the second cube
7. Pad the sketch

It should look like this:

[image]

Now resize the second cube so instead of covering the quarter it instead intersects the cube. The pad breaks:

[image]

If instead of using a BaseFeature we use a ShapeBinder and repeat the process it works. I'm guessing this is because the attachment editor maps to the fusion itself rather than the binder?

### Full version info

```shell
OS: Arch Linux (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37668 (Git) AppImage
Build type: Release
Branch: main
Hash: ea68c5e88cdc80044ede482f6ad908820f4c1c82
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * Manipulator 1.5.0
  * fasteners 0.4.67
  * Help 1.0.3
```

### Subproject(s) affected?

None

### Anything else?

[basefeature.zip](https://github.com/user-attachments/files/15755525/basefeature.zip)
[shapebinder.zip](https://github.com/user-attachments/files/15755526/shapebinder.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-11:
@bgbsww FYI

bgbsww, 2024-06-24:
During the process of turning this into a test I didn't see the problem.  But when I ran the test I definitely did.

If you turn the sketch visibility on, you can see that it isn't against the second box anymore.  I suspect this is because of a failure to correctly apply the placement of that second box.  Investigating as to why.

Here's the test:

```
def testSomething(self):
    doc = App.ActiveDocument
    doc.addObject("Part::Box","Box")
    doc.ActiveObject.Label = "Cube"
    doc.addObject("Part::Box","Box")
    doc.ActiveObject.Label = "Cube"
    from BOPTools import BOPFeatures
    bp = BOPFeatures.BOPFeatures(App.activeDocument())
    bp.make_multi_fuse(["Box", "Box001", ])
    App.activeDocument().addObject('PartDesign::Body','Body')
    doc.Body.Label = 'Body'
    App.activeDocument().Body.BaseFeature = App.activeDocument().Fusion
    doc.Body.newObject('Sketcher::SketchObject','Sketch')
    doc.Sketch.AttachmentSupport = (doc.BaseFeature,['Face2',])
    doc.Sketch.MapMode = 'FlatFace'
    geoList = []
    geoList.append(Part.LineSegment(App.Vector(11.562587, 13.382092, 0.000000),App.Vector(11.562587, 10.644792, 0.000000)))
    geoList.append(Part.LineSegment(App.Vector(11.562587, 10.644792, 0.000000),App.Vector(14.063914, 10.644792, 0.000000)))
    geoList.append(Part.LineSegment(App.Vector(14.063914, 10.644792, 0.000000),App.Vector(14.063914, 13.382092, 0.000000)))
    geoList.append(Part.LineSegment(App.Vector(14.063914, 13.382092, 0.000000),App.Vector(11.562587, 13.382092, 0.000000)))
    doc.Sketch.addGeometry(geoList,False)
    del geoList
    constraintList = []
    constraintList.append(Sketcher.Constraint('Coincident', 0, 2, 1, 1))
    constraintList.append(Sketcher.Constraint('Coincident', 1, 2, 2, 1))
    constraintList.append(Sketcher.Constraint('Coincident', 2, 2, 3, 1))
    constraintList.append(Sketcher.Constraint('Coincident', 3, 2, 0, 1))
    constraintList.append(Sketcher.Constraint('Vertical', 0))
    constraintList.append(Sketcher.Constraint('Vertical', 2))
    constraintList.append(Sketcher.Constraint('Horizontal', 1))
    constraintList.append(Sketcher.Constraint('Horizontal', 3))
    doc.Sketch.addConstraint(constraintList)
    del constraintList
    constraintList = []
    doc.recompute()
    doc.recompute()
    doc.Body.newObject('PartDesign::Pad','Pad')
    doc.Pad.Profile = (doc.Sketch, ['',])
    doc.Pad.Length = 10
    doc.recompute()
    doc.Pad.ReferenceAxis = (doc.Sketch,['N_Axis'])
    doc.Sketch.Visibility = False
    doc.Pad.Length = 10.000000
    doc.Pad.TaperAngle = 0.000000
    doc.Pad.UseCustomVector = 0
    doc.Pad.Direction = (0, -1, 0)
    doc.Pad.ReferenceAxis = (doc.Sketch, ['N_Axis'])
    doc.Pad.AlongSketchNormal = 1
    doc.Pad.Type = 0
    doc.Pad.UpToFace = None
    doc.Pad.Reversed = 0
    doc.Pad.Midplane = 0
    doc.Pad.Offset = 0
    doc.BaseFeature.Visibility = False
    doc.Sketch.Visibility = False
    # Act
    App.getDocument("Unnamed").Box001.Width='3.00 mm'
    App.getDocument("Unnamed").Box001.Height='10.00 mm'
    App.getDocument("Unnamed").Box001.Placement=App.Placement(App.Vector(5.00,5.00,5.00),App.Rotation(App.Vector(0.00,0.00,1.00),0.00))
    doc.recompute()

```

PR #14994 (2024-06-30): Toponaming: support base feature in body correctly.
Builds on #14536 and fixes #14577.  Cleans up a couple of old tests and adds two new ones.
closed 2024-06-30 by chennes