FreeCAD/FreeCAD #14578: The configuration table is not recalculated/recompute.

kaiwas, 2024-06-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello!
I'm trying to make a plane normal selection via a property table. After changing the selection (when I make a selection several times), automatic recalculation does not occur. And there is no signal in the view tree that the scene needs to be recalculated.
The algorithm works if I use the "Recalculate all" macro
This can only be noticed by seeing an object incorrectly located in space in the 3D window.

Demonstration of an example on video.
https://github.com/FreeCAD/FreeCAD/assets/32273833/8748e3e9-46d6-48e0-b81f-ca4cfe7c0af7

Test file is attached. Rename TXT --> FCstd
[Normal_Select.txt](https://github.com/user-attachments/files/15757458/Normal_Select.txt)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37668 (Git) AppImage
Build type: Release
Branch: main
Hash: ea68c5e88cdc80044ede482f6ad908820f4c1c82
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * OpenTheme 2024.5.16
  * DynamicData 2.62.0 (Disabled)
  * FreeCAD-NativeIFC 2023.10.27
  * FreeCAD_Info
  * boltsfc 2022.11.5 (Disabled)
  * Curves 0.6.37 (Disabled)
  * QuickMeasure 2022.10.28
  * sheetmetal 0.4.16 (Disabled)
  * PitchedRoof
  * Reinforcement (Disabled)
```

### Subproject(s) affected?

Spreadsheet

### Anything else?

Perhaps this problem is somehow related to this.
https://github.com/FreeCAD/FreeCAD/issues/9850
But in my case it works a little differently.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-06-13:
For some reason the spreadsheet needs to be recomputed a 2nd time.

krushia, 2024-06-16:
Having a problem here with Spreadsheet expressions involving App::PropertyXLink pointing to an App::LinkGroup not recomputing when the children of the LinkGroup change. In this instance it will sometimes show an error in a cell and a forced recompute doesn't seem to fix it.

In attached, see that cell C2 is showing an error for no valid reason. The cell below it uses the same addressing style and shows a value - but that one is an old value that should update if you reload. It seems to glitch out at random when a link target along the chain is touched.

Screenshot shows state of save. It sometimes recovers when file is reloaded, so who knows.
[image]

[buggySpreadsheet.FCStd.delete.this.extension.zip](https://github.com/user-attachments/files/15859261/buggySpreadsheet.FCStd.delete.this.extension.zip)

krushia, 2024-06-16:
Found out that doing a right-click refresh on A2 before right-click refresh on C2 will fix it. So it seems to be a caching issue with links in spreadsheets, and those not updating before updates of the cells relying on them.

krushia, 2024-06-16:
@realthunder left this nice comment in the code.
https://github.com/FreeCAD/FreeCAD/blob/6b84b65841cf1a906de0cde7b5238b0bd036dcf3/src/Mod/Spreadsheet/App/PropertySheet.cpp#L1475-L1479