FreeCAD/FreeCAD #14580: Failed to open ifc file in freecad o21.2: failed to compute placement etc.

GBWbim, 2024-06-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When trying to open ifc files exported from rvt, I receive several "failed to compute placement" messages then a complete freeze at 80% with "<class 'IndexError'>: tuple index out of range. These files can be successfully opened in other ifc-compatible programmes such as BlenderBim, Solibri Anyway, openifc viewer etc.. I would be grateful for any suggestion how to resolve this as the native  ifc capability would be ideal for my purpose
[image]

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-06-13:
Can you post a, preferably, simple example IFC file? Thanks.

GBWbim, 2024-06-13:
Hallo,

 

thanks for the response, my model files are c. 35mB, I assume too large to post by email direct. Is that ok via github?  – I am a complete beginner with github. Otherwise I could delete a few items with Blenderbim, but how small should the file be?

 

I thought about adding a try/exception at the offending line in importIFC.py to determine exactly the elements (by Tag) in order to interrogate them. Could that work first?

 

Regards,

GBW

GBWbim, 2024-06-13:
Hallo,

here is a primitive ifc model made in Blenderbim, together with its own error message

 

regards,

GBW

Roy-043, 2024-06-13:
You'll have to log into GitHub to post a file. You'll have to zip the IFC as well because GitHub allows only certain file types.

GBWbim, 2024-06-15:
Please see attached one of the files affected; church15
[Church15.zip](https://github.com/user-attachments/files/15846410/Church15.zip)

GBWbim, 2024-06-15:
I amended the importIFC file which now reveals potential faulty "Window" elements by Name, as attached screenshot, but doubt these are critical.
[image]

yorikvanhavre, 2024-06-25:
It open for me here:

[image]

Nice model ;)

There are a couple of apparently misplaced objects, though. I would say the difference here between yours and mine is probably a different version of IfcOpenShell.

The placement problems might be hard to debug... I would try to work with a smaller subset of the file, until we can pinpoint one entity that has placement problems

GBWbim, 2024-06-30:
Thanks, good news you could open the model. Isn't freecad freestanding, or where should look for a non-compatible version of ifcopenshell, python?

It open for me here:
image.png (view on web)

yorikvanhavre, 2024-07-01:
The ifcopenshell version you have is the one bundled with the FreeCAD installer, since you are using 0.21 it might be outdated. Could you try with a dev version from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds ?

GBWbim, 2024-07-01:
I downloaded the zipped file and copied the contents over into the freecad folder, overwriting the version 21 files (I hope that was correct), but when trying to open the ifc files it froze at 32% and “as native ifc” 95%. I guess I shall have to give there and wait for a new stable version. BTW I ran a script with ifcopenshell on Python Idle but had to install numpy in order to obtain the placement information. Should/can I install a different version of ifcopenshell into freecad?

yorikvanhavre, 2024-07-01:
> I downloaded the zipped file and copied the contents over into the freecad folder

That was quite bold :) but if it works, that's fine. Otherwise, you can always keep both versions side to side, an "installed" FreeCAD 0.21 + an unzipped dev version somewhere else.

If you can find your way into FreeCAD's folders, then you can download a copy of IfcOpenShell from https://docs.ifcopenshell.org/ifcopenshell-python/installation.html and unzip it where there should already be an IfcOpenShell somewhere. When downloading, make sure you take the same Python version as FreeCAD

GBWbim, 2024-07-02:
In the (newly installed) freecad 0.21.2, python console, ifcopenshell.version gives 0.6.0b0 and sys.version gives 3.8.17

In freecad 0.22dev python console ifcopenshell.version gives 0.7.0 and sys.version gives 3.11.9 packaged by conda

Could I ask what respective versions are given on your freecad version that successfully opens the church15.ifc file.

 

I see that ifcopenshell currently offers 0.7.0 for python 3.11 so is that going to help?

Can I assume there can be no conflict with the python versions 3.8 and 3.12 installed on the computer?

GBWbim, 2024-07-02:
I downloaded and unzipped build 38001 and it opened church15.ifc (at least) on an English language system (Win10x64 laptop), so I repeated this on my German Win10x64 workstation, switching the programme language to English too-it opened again! This is very encouraging and satisfactory for now – I shall be testing other files and learning the ropes. Many thanks.

yorikvanhavre, 2024-07-03:
Yes, definitely ifcopenshell v0.6.0 is now obsolete. Even v0.7.0 has been there for some time so there are improvements in its latest packages that are not in the earlier ones... It's complicated to track because they don't update the third number. Hopefully the more ifcopenshell is used in other software the more the awareness raises and these things get better.

Glad you have it working!

I guess we can close this issue as 1) it works in the current development version and 2) we don't plan to upgrade 0.21 anymore since 1.0 will be out soon. I'll close the issue then, please reopen if I missed something! (Also don't hesitate to open new bug reports if you bump into other problems - this file is quite large and I suspect you'll have situations not many people had up to now)

closed 2024-07-03 by yorikvanhavre

GBWbim, 2024-07-03:
Sorry, a final question: does the link in my particular posting as I uploaded the file mean this is in the public domain? I would prefer that it is not. If so can I delete it?

yorikvanhavre, 2024-07-03:
It is public, anyone can take it, but it's not public domain. You keep the copyright. But of course you can delete it. You can edit your message on https://github.com/FreeCAD/FreeCAD/issues/14580 and delete the links