FreeCAD/FreeCAD #14583: Example ArchDetail.FCStd cannot be opened without errors

Roy-043, 2024-06-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

ArchDetail.FCStd (,../data/examples/ArchDetail.FCStd) cannot be opened in V0.22 without errors. It was created in a version before V0.17.

@yorikvanhavre Are you OK with this?

### Full version info

```shell
V0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-06-14:
I happened to be looking at this file as an example to display 3D dimensions on a TechDraw page.

Looks like the main issue is that the technical drawing page was created with the old Drawing workbench, which is not supported anymore and was replaced by the TechDraw workbench. Without any changes to the file, the technical drawing page appears empty.

1. With these changes to the `Document.xml` file, removing the old Drawing workbench page, adding a new TechDraw page ad moving all views into it, most of the errors go away:

```diff
--- orig/Document.xml	2024-06-14 14:59:43.566501851 +0200
+++ ../ArchDetailExtract/Document.xml	2024-06-14 14:18:53.844479818 +0200
@@ -109,16 +109,16 @@
         <Object type="App::DocumentObjectGroup" name="Group" />
         <Object type="Part::FeaturePython" name="Structure023" />
         <Object type="Part::FeaturePython" name="Structure024" />
-        <Object type="Drawing::FeaturePage" name="Page" />
+        <Object type="TechDraw::DrawPage" name="Page" />
         <Object type="App::DocumentObjectGroup" name="Group001" />
         <Object type="App::FeaturePython" name="Section" />
-        <Object type="Drawing::FeatureViewPython" name="View" />
+        <Object type="TechDraw::DrawViewArch" name="View" />
         <Object type="App::FeaturePython" name="Section001" />
-        <Object type="Drawing::FeatureViewPython" name="View001" />
-        <Object type="Drawing::FeatureViewPython" name="View002" />
-        <Object type="Drawing::FeatureViewAnnotation" name="Annotation" />
-        <Object type="Drawing::FeatureViewAnnotation" name="Annotation001" />
-        <Object type="Drawing::FeatureViewAnnotation" name="Annotation002" />
+        <Object type="TechDraw::DrawViewArch" name="View001" />
+        <Object type="TechDraw::DrawViewArch" name="View002" />
+        <Object type="TechDraw::DrawViewAnnotation" name="Annotation" />
+        <Object type="TechDraw::DrawViewAnnotation" name="Annotation001" />
+        <Object type="TechDraw::DrawViewAnnotation" name="Annotation002" />
         <Object type="App::FeaturePython" name="Section002" />
         <Object type="App::FeaturePython" name="Dimension" />
         <Object type="App::FeaturePython" name="Dimension001" />
@@ -129,7 +129,7 @@
         <Object type="App::FeaturePython" name="Dimension006" />
         <Object type="App::FeaturePython" name="Dimension007" />
         <Object type="App::DocumentObjectGroup" name="Group002" />
-        <Object type="Drawing::FeatureViewPython" name="ViewGroup002" />
+        <Object type="TechDraw::DrawViewDraft" name="ViewGroup002" />
         <Object type="App::DocumentObjectGroup" name="Group003" />
         <Object type="App::FeaturePython" name="Dimension008" />
         <Object type="App::FeaturePython" name="Dimension009" />
@@ -139,7 +139,7 @@
         <Object type="App::FeaturePython" name="Dimension013" />
         <Object type="App::FeaturePython" name="Dimension014" />
         <Object type="App::DocumentObjectGroup" name="Group004" />
-        <Object type="Drawing::FeatureViewPython" name="ViewGroup004" />
+        <Object type="TechDraw::DrawViewDraft" name="ViewGroup004" />
         <Object type="App::DocumentObjectGroup" name="Group005" />
     </Objects>
     <ObjectData Count="91">
```

Resulting file with the minimum changes: 
[ArchDetail_1-0.zip](https://github.com/user-attachments/files/15838613/ArchDetail_1-0.zip)

2. This error still remains. I'm not sure where it comes from:

```
14:45:59  SketchObject::onChanged(): Unmanaged change of Constraint Property results in invalid constraint indices
14:45:59  comparison result: 1
```

3. With TechDraw's new autosnap feature for views (can it be temporarily disabled?), I can't find a way to align the the dimensions views to their correspoinding section views.

4. It might be worth adding the https://github.com/FreeCAD/FreeCAD/labels/WB%20BIM label to this bug to keep all the BIM issues together

Roy-043, 2024-06-16:
Kudos for trying this. :+1: 

But for me this is not the answer. The file would still be old and not showcase the current version. The same may apply to the other example files. The whole idea of including example files is great, but they should then be up-to-date and also exemplary. Otherwise they defeat their purpose.

PR #14937 (2024-07-02): BIM: new example file
Removed the old ArchDetail example, that was way too old (made before FreeCAD even had units!) and added a new more modern example file: 

[image]

Fixes #14583
commit 96803f34 (2024-07-02): BIM: new example file - fixes #14583
closed 2024-07-02 by yorikvanhavre