FreeCAD/FreeCAD #14584: BIM: Layer Manager code also in Draft WB

Roy-043, 2024-06-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

During this dev cycle the Layer Manager code was integrated in the Draft WB. It should be removed from the BIM WB code.

### Full version info

```shell
V0.22
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-06-20:
Note that the BIM version has a small enhancement, it adds nativeIFC functionality. But indeed that could maybe be merged in the Draft code, and used or not depending on the case. I'll have a look

maxwxyz, 2024-06-29:
maybe related: 
- https://github.com/FreeCAD/FreeCAD/issues/15038

Roy-043, 2025-07-03:
Still an issue.